rbdl/rbdl #3: Migrate to Catch2 testing framework

martinfelis, 2020-03-22:


DanHauer, 2020-06-18:
Dear Developer

Recently, I moved all of the tests to the Catch2 testing framework. In some files I "detected" some normal Cpp-Assertions which could also be realized via Catch2. 

For instance, the file [LoopConstraintsTests.cc](https://github.com/rbdl/rbdl/blob/master/tests/LoopConstraintsTests.cc) contains more than 50 assertions (first in line 437). But there are more files (not all) using assertions.

Shall I move them, too? 
Or does using these assertions have a "deeper sense"? 
Or does it depend on each individual case?

Many thanks in advance.

Best,
Daniel

martinfelis, 2020-06-28:
Hi Daniel,

thanks for your work. I think it is safe to convert them to Catch2 assertions and where not possible leave them as regular asserts. 

Regular asserts simply abort the tests right away and are not present and therefore checked in a Release build.

Please issue a pull request when you are satisfied with your work. We will then review the code.

Best regards,
Martin

closed 2021-10-16 by martinfelis