rbdl/rbdl #8: newer lua library not supported

DanHauer, 2020-03-30:
Dear Developers

I have installed Lua 5.3 via `sudo apt install liblua5.3-0 liblua5.3-dev`.

Then activating RBDL_BUILD_ADDON_LUAMODEL in `~rbdl/build$ ccmake .` leads to the following error message, probably because LUA Version 5.1 is expected:

CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
   Could NOT find Lua51 (missing: LUA_LIBRARIES LUA_INCLUDE_DIR)
 Call Stack (most recent call first):
   /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
   /usr/share/cmake-3.10/Modules/FindLua51.cmake:84 
(FIND_PACKAGE_HANDLE_STANDARD_ARGS)
   addons/luamodel/CMakeLists.txt:9 (FIND_PACKAGE)

After exitting the error message both of these entries can be set manually:
 LUA_INCLUDE_DIR                  LUA_INCLUDE_DIR-NOTFOUND                                                                                                                                                 
 LUA_LIBRARY                      LUA_LIBRARY-NOTFOUND 

It can be fixed by setting manually the correct path to lua5.3 for both entries. Then, LUA_LIBRARIES and LUA_MATH_LIBRARY are adapted automatically (Configuring twice):

 LUA_INCLUDE_DIR                  /usr/include/lua5.3                                                                                                                                                      
 LUA_LIBRARIES                    /usr/lib/x86_64-linux-gnu/liblua5.3.so;/usr/lib/x86_64-linux-gnu/libm.so                                                                                                 
 LUA_LIBRARY                      /usr/lib/x86_64-linux-gnu/liblua5.3.so                                                                                                                                   
 LUA_MATH_LIBRARY                 /usr/lib/x86_64-linux-gnu/libm.so

Since newer Lua versions are supported by RBDL, it would be great if RBDL would find them automatically as well. 

Thanks,
Daniel

mkudruss, 2020-03-31:
Hi Daniel,

thanks for sharing this with us. I think that the problem lies in addons/luamodel/CMakeLists.txt.
Would you mind trying out if you find a more general lookup than FindPackage(Lua51 Required) that works for you?

If you find a fix, then just create a pull request (PR) that we can have a look at it.

Best
Manuel

paulmanns, 2020-04-14:
Dear Developers,

Changing FIND_PACKAGE (Lua51 REQUIRED) to FIND_PACKAGE (Lua 5.1 REQUIRED) in addons/luamodel/CMakeLists.txt resolves the issue and then newer versions are also accepted on Daniel's and my systems. However, FindLua.cmake, which is required for this purpose is part of cmake from cmake version 3.0.2 on (https://cmake.org/cmake/help/v3.0/module/FindLua.html).

Suggested fix: we add the FindLua.cmake into the directory CMake/ and change the respective line of code in addons/luamodel/CMakeLists.txt. 

Best
Paul

martinfelis, 2020-05-17:
Please note I've added issue #15 to bump required CMake version to >= 3.0 so once that is fixed the files will not be necessary anymore.

closed 2020-08-17 by martinfelis