rbdl/rbdl #9: base link named as ''base_link'' for floating base robot error

xinsongyan, 2020-04-04:
I just happened to tested a robot with a its base link named as ''base_link''. 
When this robot has been loaded as a floating base robot, everything calculated are with repect to the robot base instead of the inertia base (world). Then I tried to change ''base_link'' to ''base'', everything works. I think another ''base_link'' is created as the world link name, it conflicts with the robot root link name.

martinfelis, 2020-04-05:
The parser in 2.6.0 makes some unfortunate assumptions. Can you try the development version in the dev branch?

martinfelis, 2020-04-13:
Hey @xinsongyan, does more recent code fix your issue?

xinsongyan, 2020-04-13:
Hi, @martinfelis, I haven't tried yet. I used other name to avoid the problem. I will test it later.

martinfelis, 2020-06-28:
Hey @xinsongyan, any updates on this?

martinfelis, 2020-08-24:
@xinsongyan feel free to re-open should you still have an issue.

closed 2020-08-24 by martinfelis