rbdl/rbdl #13: Python InverseDynamics does not take in f_ext

jdalmeida, 2020-05-06:
I am trying to use the python wrapper and I want to run InverseDynamics with the optional f_ext, however I get the error that InverseDynamics only takes in 5 objects. Looking in the file rbdl-wrapper.pyx it seems that python has no optional variable for f_ext and when its passed to the C++ function it is always NULL. 

[image]

mkudruss, 2020-05-17:
Hey jdalmeida,
thanks for mentioning this. I'll check it out this week and give you feedback.
Best
Manuel

mkudruss, 2020-05-25:
Hi jdalmeida,
I'll implemented the fixes but some more time as have some problems with the cython compilation.
Best
Manuel