rbdl/rbdl #16: Gravitational component of dyamic model

r1f1, 2020-06-04:
I'm finishing a task for the master I'm doing and I need this information as soon as possible. I have to create a PD control loop with gravity compensation (i.e. g(q)), I have read that it can be obtained through the InverseDynamics function and also with NonlinearEffects, but I do not understand how it is done.

mkudruss, 2020-06-06:
Hi there,
I hope you understand that we cannot help you solving the problems related to your master thesis.
However, we can maybe give you some directions. Have you looked at the documentation for said algorithms (InverseDynamics/NonlinearEffects) [here](https://rbdl.github.io)?
RNEA solves the following problem given q, qdot, qddot it computes generalized joint forces, i.e.,
  RNEA(q, qdot, qddot) = M(q)qddot + n(q, qdot) = tau
For example, NonlinearEffects (tailored RNEA for qddot = 0), computes the term n(q, qdot). It accounts for the Coriolis and centrifugal forces, gravity, and any other forces acting on the system other than those in tau. If you now play around with a simple example you should be able to find out how to compute the gravity term g(q) using NonlinearEffects.
Maybe @martinfelis or @mjhmilla can also comment on it.
Best
Manuel

r1f1, 2020-06-08:
Ok, I already understood how the algorithm works. If I only give position values ​​the value of tau is equal to the gravitational component, don't? 
And I should clarify that I think I expressed myself wrong, this is for a task of a subject of the master, not for the thesis :sweat_smile:

mkudruss, 2020-06-28:
I think that the question is answered.
Thanks for everyone's feedback.
Good luck with your masters.

closed 2020-06-28 by mkudruss