rbdl/rbdl #17: Static library build fails

alaurenzi, 2020-06-05:
I am using tag `v2.5.0` on ubuntu 16.04, with the following cmake options

```
//Build the benchmarking tool
RBDL_BUILD_ADDON_BENCHMARK:BOOL=OFF

//Build the lua model reader
RBDL_BUILD_ADDON_LUAMODEL:BOOL=OFF

//Build the (experimental) urdf reader
RBDL_BUILD_ADDON_URDFREADER:BOOL=ON

//Build experimental python wrapper
RBDL_BUILD_PYTHON_WRAPPER:BOOL=OFF

//Build statically linked library (otherwise dynamiclly linked)
RBDL_BUILD_STATIC:BOOL=ON

//Build the test executables
RBDL_BUILD_TESTS:BOOL=OFF

//Enable logging (warning: major impact on performance!)
RBDL_ENABLE_LOGGING:BOOL=OFF

//Value Computed by CMake
RBDL_SOURCE_DIR:STATIC=/home/arturo/Code/advr-superbuild/external/rbdl

//Enable storing of version information in the library (requires
// build from valid repository)
RBDL_STORE_VERSION:BOOL=OFF

//Use the URDF library provided by ROS
RBDL_USE_ROS_URDF_LIBRARY:BOOL=ON

//Use slow math instead of the fast Eigen3 library (faster compilation)
RBDL_USE_SIMPLE_MATH:BOOL=OFF
```

the linking error suggests that the static build lacks some linker flags!

```
[100%] Linking CXX executable rbdl_urdfreader_util
librbdl_urdfreader.a(urdfreader.cc.o): In function `RigidBodyDynamics::Addons::URDFReadFromString(char const*, RigidBodyDynamics::Model*, bool, bool)':
urdfreader.cc:(.text+0x39af): undefined reference to `urdf::parseURDF(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
addons/urdfreader/CMakeFiles/rbdl_urdfreader_util.dir/build.make:96: recipe for target 'addons/urdfreader/rbdl_urdfreader_util' failed
make[2]: *** [addons/urdfreader/rbdl_urdfreader_util] Error 1
CMakeFiles/Makefile2:499: recipe for target 'addons/urdfreader/CMakeFiles/rbdl_urdfreader_util.dir/all' failed
make[1]: *** [addons/urdfreader/CMakeFiles/rbdl_urdfreader_util.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

mkudruss, 2020-06-06:
Hi Arturo,
thanks for sharing this information and the . Is there any specific reason to you 2.5.0?
Is the error present when you are using master? Are you using the ROS URDF parser?
I'm just trying to reproduce your problems.
Special thanks for the remark on `-fPIC`. Will incorporate this.
Best
Manuel

mkudruss, 2020-06-06:
> Are you using the ROS URDF parser?
Just seeing that you are using ROS URDF reader :smile:

ju6ge, 2022-07-20:
Closing this issue due to inactivity after version upgrade to 3.2.0. Static build works for me on version 3.2.0 with addons enabled.

Older versions will not get bug fixes.