rbdl/rbdl #19: I have a closed loop robot. Can you give me method?

zapplelove, 2020-07-30:
I have a closed loop robot description file which is writed by SDFormat. And I want convert it to Lua file automatically, can you give me a method?

mjhmilla, 2020-08-24:
We currently do not have a method to convert a model from an SDFormat to a Lua model. If you carefully look at the constrained double pendulum example (link below) in rbdl-orb you should be able to convert the model by hand.

https://github.com/ORB-HD/rbdl-orb/tree/master/examples/constrainedDoublePendulum

zapplelove, 2020-10-29:
@mjhmilla Thanks. I think my robot model is difficult to convert by hand. I really want to find an automation method to do this.

zapplelove, 2020-10-30:
@mjhmilla I can't understand why there is a loop chain in this model.
[image]

The structure of my robot leg is like this, it's a parallelogram structure and actuated by two hydraulic cylinders at one point.
[image]

mjhmilla, 2020-11-03:
Hello @zapplelove,

While closed-loop chains can be modelled in RBDL the interface to read in the lua code for a system with loop constraints is not in the main repository. However, if you have a look at [rbdl-orb](https://github.com/ORB-HD/rbdl-orb) you will find that there are lua commands you can use to add a loop constraint to a system. If you look at the [constrained double pendulum example](https://github.com/ORB-HD/rbdl-orb/blob/master/examples/constrainedDoublePendulum/model/constrainedDoublePendulum.lua) you can see an example of how to do this. Assuming that your model can be described using bodies, joints, loop-constraints, and point constraints you can model it in lua which rbdl-orb can read in.

However, it would still be up to you (or someone) to write the code to translate from a SDFormat to lua. Everyone who works on this project does so in their spare time. Thus if you want this method, you will have to write it yourself.

ju6ge, 2022-07-20:
Closing due to inactivity.

closed 2022-07-20 by ju6ge