rbdl/rbdl #21: Get coriolis matrix value

ad1t7a, 2020-09-22:
Is it possible to get the coriolis matrix for a given model? I realize that there is no direct API call, but I wanted to calculate Ctranspose * qdot for my work. Is there a way to get this value?

martinfelis, 2020-09-27:
Can you specify what do you mean by coriolis matrix? Do you maybe mean the vector returned by NonlinearEffects (q, qdot, f_ext) (https://rbdl.github.io/d6/d63/group__dynamics__group.html#ga1782fbfb223812259d73fec58deff94e)?

thanhtam-h, 2021-02-04:
Martin, the NonlinearEffects term (N) in rbdl library can be expressed as: N = C(q,qdot)*qdot
What he meant is, how to get only C(q,qdot) separately. I think that he want to use it to estimate momentum of a robot manipulator for collision detection.

chinahuangyong, 2022-08-01:
> Martin, the NonlinearEffects term (N) in rbdl library can be expressed as: N = C(q,qdot)*qdot What he meant is, how to get only C(q,qdot) separately. I think that he want to use it to estimate momentum of a robot manipulator for collision detection.

i also meet this problem, solved it?

DaegyuLim, 2022-12-21:
One of the easiest ways to circumvent this problem is by using '-Mdot*qdot + C*qdot' instead of 'Ctranspose * qdot'. Mdot can be obtained from simple numerical derivatives. C*qdot is equal to the NonlinearEffects API values. But this method produces some errors on Mdot especially when the velocity is fast. C matrix also can be obtained from the inertia matrix M using the Christoffel Symbol method. But this method costs a lot of computation if the DoF of the model is large (maybe O(N^3)). I'm trying to implement a recent work presented in this paper (https://arxiv.org/abs/2010.01033) which offers MATLAB code(https://github.com/ROAM-Lab-ND/spatial_v2_extended).

Xunmenggod, 2024-12-17:
> matrix M using the

Hi, I'd like to achieve a momentum-based observer. Is there a way to get the C matrix directly?