rbdl/rbdl #24: Add integration to 📦 VCPKG

RDCH106, 2020-10-06:
Add port for [VCPKG](https://github.com/microsoft/vcpkg) package management tool.

Current PR waiting for integration: https://github.com/microsoft/vcpkg/pull/13884

⚠️ If PR is accepted, the README must be updated

RDCH106, 2020-10-13:
RBDL was included in VCPKG: https://github.com/microsoft/vcpkg/pull/13884#pullrequestreview-506244495

🎉🎉🎉

And it will be available for the next release.

RDCH106, 2020-10-13:
⚠️⏳ Keep open this issue until the usage of the RBDL library with VCPKG will be documented in README

martinfelis, 2020-10-21:
Hi, thanks for the contribution. This is really cool! 

Can you create a pull request for what exactly you would like to have in README? I have not yet used VCPKG so would greatly appreciate more specific details.

RDCH106, 2020-10-22:
Sure!!

I will generate the specific documentation in the README in a PR when the next release of [vcpkg](https://github.com/microsoft/vcpkg) is ready.

It will improve the usability and distribution of the RBDL library. RBDL is prepared by vpkg using the release tags of this repository.

commit 70ceef9d (2021-05-16): Merge pull request #28 from RDCH106/vcpkg

Add VCPKG section to README.md (fix #24)
commit f06fb854 (2021-05-16): Add VCPKG section to README.md (fix #24)
closed 2021-05-16 by martinfelis