rbdl/rbdl #29: Python wrapper and undefined `ext_modules`

mahaarbo, 2021-03-20:
Hi,

I think something may have gone missing in the transition to `python3`. 
When trying to install it now I get `NameError: name 'ext_modules' is not defined`. I can see that it is defined here https://github.com/rbdl/rbdl/blob/242bf36fbae13ef2b67414e23844f59f97d24ea1/python/setup.py.cmake#L28
 
I think something went wrong when dev was merged into master as the the [last commit](https://github.com/rbdl/rbdl/blob/1caa4b73c1322a504d5f4eb3fdba537dad5ccdff/python/setup.py.cmake) by ju6ge did not have `ext_modules=cythonize(ext_modules)` in it.
But simply removing the line with `ext_modules` does not solve the problem as I get:
`undefined symbol: _ZN17RigidBodyDynamics6Addons16URDFReadFromFileEPKcPNS_5ModelEbb`
Which may be un
I am building with `RBDL_BUILD_ADDON_URDFREADER` and `RBDL_BUILD_PYTHON_WRAPPER` on ubuntu 20.04.

YummyOu, 2021-04-22:
I also got this problem!
I use python2 on ubuntu 16.04.

Vishnuparammal, 2021-05-04:
I got the same error when compiling with these commands on Ubuntu 18.04 LTS 
`cmake -D CMAKE_BUILD_TYPE=Release -D RBDL_BUILD_ADDON_URDFREADER=ON -D RBDL_BUILD_PYTHON_WRAPPER=ON ../`
`make -j8`
`sudo make install`

Vishnuparammal, 2021-05-04:
I too tried to compile by removing the line  `ext_modules=cythonize(ext_modules)` . I had to make some changes like adding Eigen/Dense to my rbdl/include directory after which the code compiled and installed successfully for python3.
But on importing rbdl from python3, it gave segmentation fault (core dumped) error. Any leads as to why this might have occurred would be helpful.

ju6ge, 2021-05-04:
Hey guys,

I can not really tell what is going on here. The commits from me on this repository are quite old and were submitted when RBDL was still hosted on Bitbucket. Since then a lot has happened, mainly the version of RBDL I am am working on has significantly diverged from this version. 

I am developing on RBDL as part of my work at the [ORB](https://github.com/orb-hd) robotics group at the university of Heidelberg. We needed to update a lot of things and sadly our changes where never merged, some due to differences in opinion and others out of a lack of time for properly reviewing them. This has lead to a second version of RBDL which can be found here [rbdl-orb](https://github.com/orb-hd/rbdl-orb). Most things there are still the same, and from a user standpoint it should not matter which version you are using. And for the version there i can attest to the python wrapper working ;)

@mjhmilla  and I  are quite frustrated by the current state of their being two rbdl versions. There is also mention of this in PR #14, but as far as I am aware their has not been any public communication on how or if this can/will be resolved. I also don't get the impression that @mfelis or @mkudruss have time to deal with this.

For me this means that i will keep working on [rbdl-orb](https://github.com/orb-hd/rbdl-orb) and [rbdl-toolkit](https://github.com/orb-hd/rbdl-toolkit) because at least their I can get things done and it is the software that is used by our research group and it will be referenced in our future publications due to the new stuff not being present on this repository. 

Kind regards,
ju6ge

mkudruss, 2021-05-05:
Hi everyone,

@vishnu do you mind creating a PR.
If I find time, I can investigate it then.
In the meanwhile, you could try building with DEBUG and run gdb on top of
the python interpreter. The python implementation is rather experimental at
the moment but can be extended.

Best
Manuel

visonpon, 2021-07-13:
Hi，guys, I also encountered `NameError: name 'ext_modules' is not defined` when I use` python setup.py install`, has anyone got some solutions? thanks~

swframe, 2021-09-29:
y-all need to google more :)

I google the cython docs for what ext_modules should be set to.
Adding this to the PATH_TO_YOUR_BUILD_DIR/python/setup.py file should fix the problem.
    ext_modules="PATH_TO_YOUR_BUILD_DIR/python/rbdl.pyx"
There is a way to config cmake to do this but I don't have time to investigate that path.

Note, for me, the 'make install' didn't work for these reasons:

1. The setup.pyx file is missing a declaration to tell cython that the language is c++; I added '# distutils: language = c++' to the generated pyx file but this should be added the source file (which I didn't do because I'm lazy; sorry). See https://cython.readthedocs.io/en/latest/src/userguide/wrapping_CPlusPlus.html
2. cython defaults to python2 so I had to add compiler_directives={'language_level' : "3"}. See https://stackoverflow.com/questions/34603628/how-to-specify-python-3-source-in-cythons-setup-py
3. The gcc include path was missing a lot of things; (again, this should be fixed by the cmake config). 
4. This is the embarrassingly silly thing I did instead: (note my conda env is named 'CONDA_ENV')
ln -s ~/miniconda3/envs/CONDA_ENV/lib/python3.7/site-packages/numpy/core/include/numpy ~/miniconda3/envs/CONDA_ENV/include/python3.7m/
ln -s /usr/local/include/eigen3/* ~/miniconda3/envs/CONDA_ENV/include/python3.7m/
ln -s PATH_TO_RBDL_DIR/python/*.h ~/miniconda3/envs/CONDA_ENV/include/python3.7m/
ln -s PATH_TO_RBDL_DIR/python/*.cc ~/miniconda3/envs/CONDA_ENV/include/python3.7m/

This gave me clean build but I have not yet tested it so please forgive me if there are additional steps I missed.
I will update this issue later if I find anything else I needed.

Timkeeper2018, 2022-05-07:
@swframe 
I tried the method you suggested, but there are still some problems:
"rbdl_loadmodel.cc:8:10: fatal error: addons/urdfreader/urdfreader.h"
does not have that file or directory
Do you have a solution please?

Timkeeper2018, 2022-05-07:
> 你们都需要谷歌更多:)
> 
> 我用谷歌搜索cython文档，看看应该设置什么ext_modules。将其添加到PATH_TO_YOUR_BUILD_DIR/python/setup.py文件应该可以解决问题。 ext_modules=“PATH_TO_YOUR_BUILD_DIR/python/rbdl.pyx” 有一种方法可以配置 cmake 来执行此操作，但我没有时间调查该路径。
> 
> 请注意，对我来说，由于以下原因，“进行安装”不起作用：
> 
> 1. rbdl.pyx 文件缺少一个声明，用于告诉 cython 该语言是 c++;我在生成的pyx文件中添加了'# distutils： language = c ++'，但这应该添加到源文件中（我没有这样做，因为我懒惰;对不起）。查看 https://cython.readthedocs.io/en/latest/src/userguide/wrapping_CPlusPlus.html
> 2. cython默认为python2，所以我不得不在 setup.py 中添加compiler_directives={'language_level' ： “3”}。请参见 https://stackoverflow.com/questions/34603628/how-to-specify-python-3-source-in-cythons-setup-py
> 3. gcc包含路径缺少很多东西;（同样，这应该由 cmake 配置修复）。
> 4. 这是我做的令人尴尬的愚蠢的事情：（注意我的conda env被命名为“CONDA_ENV”） ln -s ~/miniconda3/envs/CONDA_ENV/lib/python3.7/site-packages/numpy/core/include/numpy ~/miniconda3/envs/CONDA_ENV/include/python3.7m/ ln -s /usr/local/include/eigen3/* ~/miniconda3/envs/CONDA_ENV/include/python3.7m/ ln -s PATH_TO_RBDL_DIR/python/*.h ~/miniconda3/envs/CONDA_ENV/include/python3.7m/ ln -s PATH_TO_RBDL_DIR/python/*.cc ~/miniconda3/envs/CONDA_ENV/include/python3.7m/
> 
> 这给了我干净的构建，但我还没有测试它，所以如果我错过了其他步骤，请原谅我。如果我找到我需要的任何其他内容，我稍后会更新此问题。

I tried the method you suggested, but there are still some problems:
"rbdl_loadmodel.cc:8:10: fatal error: addons/urdfreader/urdfreader.h"
does not have that file or directory
Do you have a solution please?

Timkeeper2018, 2022-05-07:
> y-all need to google more :)
> 
> I google the cython docs for what ext_modules should be set to. Adding this to the PATH_TO_YOUR_BUILD_DIR/python/setup.py file should fix the problem. ext_modules="PATH_TO_YOUR_BUILD_DIR/python/rbdl.pyx" There is a way to config cmake to do this but I don't have time to investigate that path.
> 
> Note, for me, the 'make install' didn't work for these reasons:
> 
> 1. The rbdl.pyx file is missing a declaration to tell cython that the language is c++; I added '# distutils: language = c++' to the generated pyx file but this should be added the source file (which I didn't do because I'm lazy; sorry). See https://cython.readthedocs.io/en/latest/src/userguide/wrapping_CPlusPlus.html
> 2. cython defaults to python2 so I had to add compiler_directives={'language_level' : "3"} to setup.py. See https://stackoverflow.com/questions/34603628/how-to-specify-python-3-source-in-cythons-setup-py
> 3. The gcc include path was missing a lot of things; (again, this should be fixed by the cmake config).
> 4. This is the embarrassingly silly thing I did instead: (note my conda env is named 'CONDA_ENV')
>    ln -s ~/miniconda3/envs/CONDA_ENV/lib/python3.7/site-packages/numpy/core/include/numpy ~/miniconda3/envs/CONDA_ENV/include/python3.7m/
>    ln -s /usr/local/include/eigen3/* ~/miniconda3/envs/CONDA_ENV/include/python3.7m/
>    ln -s PATH_TO_RBDL_DIR/python/*.h ~/miniconda3/envs/CONDA_ENV/include/python3.7m/
>    ln -s PATH_TO_RBDL_DIR/python/*.cc ~/miniconda3/envs/CONDA_ENV/include/python3.7m/
> 
> This gave me clean build but I have not yet tested it so please forgive me if there are additional steps I missed. I will update this issue later if I find anything else I needed.

When I try to install with the RBDL_BUILD_ADDON_URDFREADER flag turned on, the installer prompts me that the file cannot be found "rbdl_loadmodel.cc:8:10: fatal error: addons/urdfreader/urdfreader.h: no such file"
Do you know the solution please?

ju6ge, 2022-07-20:
Closing after version upgrade to rbdl 3.2.0. Any further issues with the python build should be handled in new issues.

closed 2022-07-20 by ju6ge