rbdl/rbdl #31: Boost errors when compiling with the URDF_READER add-on

alessandropalleschi, 2021-05-20:
Hi, 
I am having some issues when compiling the library with the URDF_READER add-on.
Using the make command the following boost-related errors arise:

[ 71%] Built target rbdl
Scanning dependencies of target rbdl_urdfreader
[ 78%] Building CXX object addons/urdfreader/CMakeFiles/rbdl_urdfreader.dir/urdfreader.cc.o
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:15:9: error: ‘boost’ does not name a type; did you mean ‘bool’?
 typedef boost::shared_ptr<urdf::Link> LinkPtr;
         ^~~~~
         bool
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:16:15: error: ‘boost’ does not name a type; did you mean ‘bool’?
 typedef const boost::shared_ptr<const urdf::Link> ConstLinkPtr;
               ^~~~~
               bool
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:17:9: error: ‘boost’ does not name a type; did you mean ‘bool’?
 typedef boost::shared_ptr<urdf::Joint> JointPtr;
         ^~~~~
         bool
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:18:9: error: ‘boost’ does not name a type; did you mean ‘bool’?
 typedef boost::shared_ptr<urdf::ModelInterface> ModelPtr;
         ^~~~~
         bool
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:39:16: error: ‘LinkPtr’ was not declared in this scope
 typedef vector<LinkPtr> URDFLinkVector;
                ^~~~~~~
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:39:23: error: template argument 1 is invalid
 typedef vector<LinkPtr> URDFLinkVector;
                       ^
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:39:23: error: template argument 2 is invalid
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:40:16: error: ‘JointPtr’ was not declared in this scope
 typedef vector<JointPtr> URDFJointVector;
                ^~~~~~~~
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:40:16: note: suggested alternative: ‘Joint’
 typedef vector<JointPtr> URDFJointVector;
                ^~~~~~~~
                Joint
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:40:24: error: template argument 1 is invalid
 typedef vector<JointPtr> URDFJointVector;
                        ^
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:40:24: error: template argument 2 is invalid
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:41:21: error: ‘LinkPtr’ was not declared in this scope
 typedef map<string, LinkPtr > URDFLinkMap;
                     ^~~~~~~
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:41:29: error: template argument 2 is invalid
 typedef map<string, LinkPtr > URDFLinkMap;
                             ^
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:41:29: error: template argument 4 is invalid
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:42:21: error: ‘JointPtr’ was not declared in this scope
 typedef map<string, JointPtr > URDFJointMap;
                     ^~~~~~~~
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:42:21: note: suggested alternative: ‘Joint’
 typedef map<string, JointPtr > URDFJointMap;
                     ^~~~~~~~
                     Joint
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:42:30: error: template argument 2 is invalid
 typedef map<string, JointPtr > URDFJointMap;
                              ^
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:42:30: error: template argument 4 is invalid
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:44:42: error: ‘ModelPtr’ has not been declared
 bool construct_model (Model* rbdl_model, ModelPtr urdf_model, bool floating_base, bool verbose) {
                                          ^~~~~~~~
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc: In function ‘bool RigidBodyDynamics::Addons::construct_model(RigidBodyDynamics::Model*, int, bool, bool)’:
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:45:3: error: ‘LinkPtr’ was not declared in this scope
   LinkPtr urdf_root_link;
   ^~~~~~~
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:48:24: error: base operand of ‘->’ is not a pointer
   link_map = urdf_model->links_;
                        ^~
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:51:25: error: base operand of ‘->’ is not a pointer
[... 134 lines cut ...]
                                                                                     urdf_model
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc: In function ‘bool RigidBodyDynamics::Addons::URDFReadFromString(const char*, RigidBodyDynamics::Model*, bool, bool)’:
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:312:3: error: ‘ModelPtr’ was not declared in this scope
   ModelPtr urdf_model = urdf::parseURDF (model_xml_string);
   ^~~~~~~~
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:312:3: note: suggested alternative: ‘Model’
   ModelPtr urdf_model = urdf::parseURDF (model_xml_string);
   ^~~~~~~~
   Model
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:314:32: error: ‘urdf_model’ was not declared in this scope
   if (!construct_model (model, urdf_model, floating_base, verbose)) {
                                ^~~~~~~~~~
/home/ale-cp/rbdl/addons/urdfreader/urdfreader.cc:314:32: note: suggested alternative: ‘model’
   if (!construct_model (model, urdf_model, floating_base, verbose)) {
                                ^~~~~~~~~~
                                model
addons/urdfreader/CMakeFiles/rbdl_urdfreader.dir/build.make:62: recipe for target 'addons/urdfreader/CMakeFiles/rbdl_urdfreader.dir/urdfreader.cc.o' failed
make[2]: *** [addons/urdfreader/CMakeFiles/rbdl_urdfreader.dir/urdfreader.cc.o] Error 1
CMakeFiles/Makefile2:664: recipe for target 'addons/urdfreader/CMakeFiles/rbdl_urdfreader.dir/all' failed
make[1]: *** [addons/urdfreader/CMakeFiles/rbdl_urdfreader.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

I am using Ubuntu 18.04, with Ubuntu 16.04 I am able to compile the everything without errors.
Do you have any idea how to fix this?

graiola, 2021-06-01:
Hi, check PR #12

ju6ge, 2022-07-20:
Closing after update to version 3.2.0

closed 2022-07-20 by ju6ge