rbdl/rbdl #34: Question about Floating Base model

MyHonorHou, 2021-07-05:
Add six virtual joints on the base to handle the floating base model. When solve the inverse dynamics, what is the meaning of the results values of virtual joints. 
For example, I use RBDL function InverseDynamics(model, q, Dq, DDq, Tau) to solve inverse dynamics, and the return Tau has the values of virtual joints. Why the values of virtual joints is not equal to zero? If they must have values, what the meaning is?

martinfelis, 2021-07-09:
The resulting forces are the forces have to be applied to produce the resulting motion.

What you are probably trying to achieve is to model an underactuated system with external forces. E.g. a two-legged system with one foot on the ground and an unactuated (i.e. forces = 0) floating base.

There are two options to achieve this: 
1. Model the system as a fully actuated system such that the foot on the ground is the root and you end up with a chain to the other foot.
2. Use InverseDynamicsConstraints() from the fork in rbdl-orb at https://github.com/ORB-HD/rbdl-orb.

MyHonorHou, 2021-07-09:
> The resulting forces are the forces have to be applied to produce the resulting motion.
>
> What you are probably trying to achieve is to model an underactuated system with external forces. E.g. a [...]

MyHonorHou, 2021-07-09:
> The resulting forces are the forces have to be applied to produce the resulting motion.
>
> What you are probably trying to achieve is to model an underactuated system with external forces. E.g. a [...]

Thank you very much, I will have a try.

closed 2021-07-09 by MyHonorHou