rbdl/rbdl #35: conda package for python wrapper?

daquang, 2021-07-30:
Hi, I was wondering if there are any plans to create a conda package for rbdl, including the python wrapper? It would make installation quite a bit easier.

martinfelis, 2021-08-06:
Hi, I am not too familiar with conda, but I'd be happy to accept pull requests in this regard.

traversaro, 2021-08-06:
At least at the conda-forge level (that is a specific channel among the many you can use in conda) there seems to be a rbdl package mantained by @pariterre in https://github.com/conda-forge/rbdl-feedstock , even if at the first glance it does not seem to contain the Python bindings.

pariterre, 2021-08-06:
Hello!
The package I maintain is from the fork of orb. We added casadi to the available math back end. That said, I did not compiled the Python wrapper (which should not be hard to add actually)..
@martinfelis, if you want to give a go, even though it is a fork of RBDL and not this repo, you are very welcomed!
The files to adapt would be in recipe

traversaro, 2021-08-06:
> The package I maintain is from the fork of orb. 

Thanks for the clarification, I tought it was the original version as the website listed in the meta.yaml is the one of rbdl: https://github.com/conda-forge/rbdl-feedstock/blob/01cba91aa681830dd9ba58b45e77c15a8ceeb44f/recipe/meta.yaml#L43 .

pariterre, 2021-08-06:
Yeah you are right, I did not know if I was supposed to point toward the newer fork or the old one... And apart from a refactor and the as of casadi, they are very similar. That said, it will probably go further and further as the time goes..

ju6ge, 2022-07-20:
@pariterre given that we now merged rbdl and rbdl-orb back together. It seems that we can make your conda build more "official". I know you are constrained for time. So maybe just update the your build and downstream packages when you find the time to do it and give a little heads up here ;)

pariterre, 2022-07-20:
I opened the PR to move the conda package :)
https://github.com/conda-forge/rbdl-feedstock/pull/32