rbdl/rbdl #36: InverseKinematics solution gives joint angles outside the joint range

saikishor, 2021-09-10:
Hello,

I have been testing the new `InverseKinematics` with the `InverseKinematicsConstraintSet` (https://github.com/rbdl/rbdl/blob/master/include/rbdl/Kinematics.h#L361-L412). I am loading the RBDL from the URDF model from ROS. The end result I am getting is way beyond the joint range and some of the angles are more than 4*PI. Is this normal and intended?.

I have also observed that some of the analytically feasible configurations are not being feasible with this InverseKinematics Method. What's the success rate of this IK?. In order to increase the changes only changing the max_steps and tolerances or is there anything else that can be done?

Thank you,

martinfelis, 2021-10-16:
In this case the inverse kinematics would be a constrained optimization problem (meaning constraints on the free optimization variables q, not inverse kinematic constraints). The solver does not support this unfortunately.

As a simple start I would suggest simply clamping the values yourself after each iteration. While not mathematically sound, it may yield sufficiently good results.

saikishor, 2021-12-03:
@martinfelis Sorry to get back to you after long time. Right now, how I am clamping those values is by doing a fmod to the result by M_PI, as shown here (`fmod(Q_res(i), M_PI)`). This sometimes is resulting in good values and it works, however, time to time, still I have a situation where the [InverseKinematics](https://github.com/rbdl/rbdl/blob/master/include/rbdl/Kinematics.h#L403) returns true, but the result is no way near the actual result I am expecting. What can i do to make sure that I always get out of this local minima?.

im-renpei, 2023-06-17:
Hi @saikishor , have you found a solution to this problem? Or do you have any good suggestions?

saikishor, 2023-06-17:
> Hi @saikishor , have you found a solution to this problem? Or do you have any good suggestions?

Hello @im-renpei!

Unfortunately, I was unable to find a proper reliable solution using RBDL as @martinfelis explained sometimes it might get stuck in the local minima. So, instead I tried to switch to a different IK solver for my use case.

Thank you,

Best Regards,
Sai Kishor Kothakota