rbdl/rbdl #41: Cannot Build Add on URDFreader

samtkeen, 2021-11-19:
I am trying to build with CMake and have managed to do successfully to create a static library for rbdl by itself. I then wanted to use the urdf reader but when including the flag to build the urdf add on i got an error saying:

4>LINK : fatal error LNK1104: cannot open file 'Release\rbdl_urdfreader.lib'

is there a fix to this or can i access a built version of the urdf addon to add to my project?

ju6ge, 2021-11-23:
Hey, I have spent quite a long time getting all the addons to run on windows. If you just want to use rbdl on windows i would suggest you take a look at [vcpkg](https://vcpkg.io). It is a complete library environment that can be used together with cmake to manage all your dependencies. There are two packages there for rbdl. Since you want to use the urdf-addon I would suggest you install the `rbdl-orb` version with `vcpkg.exe install rbdl-orb`. It has the Lua-Model and URDF-Model Addon included by default ;)

samtkeen, 2022-01-22:
the vcpkg method has worked(suprisingly much easier and more robust than i had thought), thank you for your help !

closed 2022-01-22 by samtkeen