rbdl/rbdl #42: Access Read Violation on Inverse Kinematics Function

samtkeen, 2022-01-28:
When calling the InverseKinematics(Model, Qinit, ConstraintSet, Qres) I get 
"Access violation reading location 0x0000000000000000."
with no indication where the violation is?
Is there an example of using the InverseKinematics function anywhere so i can check where I am missing something?

The model is imported via URDF using the AddOn, i have defined the body ids, target rotation and location, weights and used the end effector location as body_points in the constraint set and cant find any null pointers in the variables.

ju6ge, 2022-01-28:
Hey, did you also pass an intialized Qres Vector to the function, that is where the result will be placed.

For an example of using InverseKinematics you can take a look at the [RBDL-Toolkit Inverse Kinematics Plugin](https://github.com/ORB-HD/rbdl-toolkit/blob/master/plugins/InverseKinematicsPlugin/InverseKinematicsPlugin.cc).

And just a quick note that explaining what you are doing is not as helpful as just posting the code snippet that produces the error ;). That way people can more easly see if there are any obvious mistakes ;)

samtkeen, 2022-01-28:
yes I set Qres up as a vector of zeros. Thank you for sending that link, I was trying to attempt to use the secondary version of the function which uses a constraint set and a target rotation instead though, but if i can get either version to work that would be great!

Also apologies i thought i had attached the code snippets but obviosuly missed it. I have added my attempt to use the same variation of the inverse kinematics function you have used in the toolkit which compiles and runs(see attatched zip), however it  returns false on the call for Inverse Kinematics function.

Can i check that the units should be radians for angles and meters for locations as maybe this casues issue with the functions if they arent scaled properly?

[IKFunctions.zip](https://github.com/rbdl/rbdl/files/7960739/IKFunctions.zip)