rbdl/rbdl #43: Compile error on Ubuntu-20.04

ykk648, 2022-02-08:
`Consolidate compiler generated dependencies of target rbdl
[  7%] Building CXX object CMakeFiles/rbdl.dir/src/rbdl_version.cc.o
[ 15%] Building CXX object CMakeFiles/rbdl.dir/src/rbdl_mathutils.cc.o
In file included from /workspace/codes/rbdl-master/include/rbdl/rbdl_math.h:66,
                 from /workspace/codes/rbdl-master/include/rbdl/rbdl_mathutils.h:14,
                 from /workspace/codes/rbdl-master/src/rbdl_mathutils.cc:14:
/workspace/codes/rbdl-master/include/rbdl/Quaternion.h: In member function ‘RigidBodyDynamics::Math::Quaternion RigidBodyDynamics::Math::Quaternion::slerp(double, RigidBodyDynamics::Math::Quaternion) const’:
/workspace/codes/rbdl-master/include/rbdl/Quaternion.h:77:73: error: no matching function for call to ‘RigidBodyDynamics::Math::Quaternion::sqrt(double) const’
   77 |       double sin_half_theta = sqrt(1.0 - cos_half_theta * cos_half_theta);
      |                                                                         ^
In file included from /usr/include/eigen3/Eigen/Core:436,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /workspace/codes/rbdl-master/include/rbdl/rbdl_math.h:34,
                 from /workspace/codes/rbdl-master/include/rbdl/rbdl_mathutils.h:14,
                 from /workspace/codes/rbdl-master/src/rbdl_mathutils.cc:14:
`

Cdfghglz, 2022-02-09:
Same here on 18.04

ykk648, 2022-02-09:
@Cdfghglz  try [rbdl-orb](https://github.com/ORB-HD/rbdl-orb)

closed 2022-02-09 by ykk648

Cdfghglz, 2022-02-09:
Thanks @ykk648 the orb one works.