rbdl/rbdl #53: Acceleration and jacobian for the chosen point on the body

antczakmab, 2022-05-10:
Hi,
I try to get jacobian and acceleration for end effector. I'm using methods: CalcPointJacobian and CalcPointAcceleration. I'm not sure how to properly choose point of the body for calculations. In the documentation I found that it's in the body coordinates. I can get transformation between body and base origins, but I don't know if that origin is in the body COM, or in the joint location. How it works?
Very helpful in my case would be access to coordinations of all bodies in the model. I import model from urdf, and model is simplified, so I don't have access to all of them. In the urdf I have some virtual bodies only for reference, to be able to get their position/jacobian/accelerations. But they are not in the list when I print named body origins overview, and their ID's don't exist. Is there any option to get that data?
I use RBDL-ORB version of the library.

ju6ge, 2022-05-10:
Hi,
the body point referenced in any function is always relative to the body in the bodies frame of reference. So supplying (0,0,0) always gives the value at the body root. How the frame of reference is oriented depends on how the body is attached to the model, so there is no easy answer here.
Regarding your second question about reading points from the model file. This has not been implemented for urdf models. There is however something similar available for lua-models (https://github.com/ORB-HD/rbdl-orb/blob/master/addons/luamodel/luamodel.h). The way it works is that there are different functions to retrieve different values saved in the model file. The normal load model function only constructs the model. Implementing this for the urdf model should be done in a similar fashion and I would happily merge any pull request that adds more functionality in this regard. For this to be done the [urdf parser](https://github.com/ORB-HD/URDF_Parser) needs to be extended, but that should be straight forward, because the xml parsing is handled by tinyxml. So all that is required is adding a structure to handle body points and then accessing the xml elements to read the values. Then a function can be added to the urdf addon to read the points from the model structure.

antczakmab, 2022-05-13:
Hi, thank you for the answer. 
Maybe I'll add some features to the urdf reader, but now I'd like to understand how the model is constructed from urdf. I have continuous(rotary) and fixed joints, that define translations and rotations between links. For now, I see that bodies connected by fixed joints are merged into one and it's ok. I wonder how is constructed set of transfomations between the last named body and base. If I collect and stack into one transformation all translations and rotations that define kinematic chain in URDF, I should get the same result as CalcBodyToBaseCoordinates() method, am I right? I computed that by hand, with translations and rotations, and I got difference in one axis(maybe I missed something, I'll check it again).
Another issue that I have, is with the generalized coordinates "zeros". Can I assume that the position described by transformations in the urdf define state with all generalized coordinates at their zeros? This is important for me, because I noticed, that carthesian coordinates from CalcBodyToBase Coordinates(Z axis in my case) differ from that what I expect. Also, I'm not sure for what state of the robot I compute joint space inertia matrix, forward and inverse dynamics. What can I assume as the ground truth? I compare the results with physics simulator, where I have visualisation, but it would be much better to know how robot stands without guessing.

ju6ge, 2022-05-14:
I am afraid I do not have the time to help you debug your model. Maybe it helps if you take a look at [rbdl toolkit](https://github.com/orb-hd/rbdl-toolkit). With it you can visualize how rbdl `sees` your model. It uses the CalcBodyToBaseCoordinates and Orientation functions to visualize the model in 3D Space. It can load lua and urdf models. Maybe this helps you understand what is happening in rbdl. And to be clear it is very possible that there is some bugs to be found. The URDF Parser has not recieved same amount of attention as the lua model code.

antczakmab, 2022-05-16:
Thank you for help, I will look at the toolkit.

closed 2022-05-16 by antczakmab