rbdl/rbdl #56: Is it possible to get Forces acting on floating base, from constrained Inverse Dynamics function?

Kwach00, 2022-07-13:
Hi!
I'm working on Floating Base model of quadruped. I want to get forces acting on robot's body from model and accelerations (of leg joints and acting on the body). To compute inverse dynamics I implemented `InverseDynamicsConstraints()` and `InverseDynamicsConstraintsRelaxed()` functions. Possible Constraint Sets are defined and system is switching between them depend on robot state (I read that this is now best solution for changing constraint sets). Robot is underactuated so mainly `Relaxed` function is used. But forces that I get from function have only zeroes for forces acting on floating base. Is there a possibility to get these forces? Cause I assume that there always will be forces acting on robot's body. 
I also tried `InverseDynamics()`, but computed forces sometimes are extremely big and I can't find cause of this behavior. My another idea was to "deceive" RBDL and set all `actuatedDoF` to `true` in `SetActuationMap()` method, but it crashed program. Maybe I can get those forces with another method?
Thanks in advance for your help!

ju6ge, 2022-07-13:
This is a question for mjhmilla, he has the most experience when it comes to questions like this. 

As a general note, when asking these specific questions it would be helpful if you not just describe what you are doing, but also provide a code sample. That way obvious mistakes like handling of function arguments can be seen directly ;)

Kind regards,
ju6ge

Kwach00, 2022-07-13:
Thank you so much for the quick reply!
Of course i will provide a code samples! For starters I just wanted to give you a general look on the problem I'm dealing with :)

This is implementation of inverse dynamics functions:
```
void DynamicsModel::computeInvDynConst(const std::vector<Eigen::Vector3f> &force, const Eigen::VectorXf &qddot_desired)
{
    RigidBodyDynamics::Math::VectorNd QDDotOutput; //Generalized accelerations
    RigidBodyDynamics::Math::VectorNd QDDotDesired = qddot_desired.cast<double>();
    std::vector<SpatialVector> f_ext;

    for(size_t i = 0; i<force.size(); i++)
        f_ext.push_back(convVecfromR3toSpatial(force[i]));

    if(isSystemFullyActuated(force, constraintsCurrent))
    {
        if(force.size() != 0)
            RigidBodyDynamics::InverseDynamicsConstraints(*dynModel, Q, QDot, QDDotDesired, constraintsCurrent, QDDotOutput, Tau, &f_ext);
        else
            RigidBodyDynamics::InverseDynamicsConstraints(*dynModel, Q, QDot, QDDotDesired, constraintsCurrent, QDDotOutput, Tau);
    }
    else
    {
        // QDDotDesired = QDDotControls for InverseDynamicsConstraintsRelaxed() function
        if(force.size() != 0)
            RigidBodyDynamics::InverseDynamicsConstraintsRelaxed(*dynModel, Q, QDot, QDDotDesired, constraintsCurrent, QDDotOutput, Tau, &f_ext);
        else
            RigidBodyDynamics::InverseDynamicsConstraintsRelaxed(*dynModel, Q, QDot, QDDotDesired, constraintsCurrent, QDDotOutput, Tau);
    }
}
```

mjhmilla, 2022-07-13:
Dear @Kwach00,

Here are some things you might try:

- InverseDynamics will only do what you want if you know the external forces acting on the quadruped (probably foot ground forces). Since you probably don't know these things, you'll have to try some of the special constrained options.

- Given that you want to know the forces that the leg is applying to the body, you have some work to do. I am going to assume that you have a desired q, qdot, and qddot that you would like the model to follow. I am also going to assume that some feet are on the ground ...

1. You will have to add a constraint between the each foot and the ground. I'd recommend a point-normal constraint in 3 directions. You will need a different constraint set each time the contact set changes. Depending on the gait you are looking at, you might have as few as 2 different phases (bi laterally symmetric gallop) or many more for something like a Tölt.
2. Then use InverseDynamicsConstraintsRelaxed, or InverseDynamicsConstraints as appropriate.
3. You will then get a vector of generalized forces. You won't get the forces the leg applies to the body, but you will get the generalized force of every joint. This might be what you want.

Two closing points:
- If you really want the forces that the leg applies to the body, you will have to convert one of the joints (say the hip joint) to its equivalent representation as a constraint (a 1 dof pin joint can be represented as 5 constraints). Then the reaction forces will show up as constraint forces, and you can use functions in RBDL's Constraints library to get at these values.

- When you manually changed the SetActuationMap there is a good chance you over-constrained the model. Neither InverseDynamicsConstraintsRelaxed nor InverseDynamicsConstraints can handle over-constrained systems. It is sometimes not trivial to figure out if a model is over constrained or not, since this can depend on the geometry of the model and its current state.

Kwach00, 2022-07-15:
Thank you @mjhmilla for explanation!

I actually know external forces from mpc. So maybe `InverseDynamics` would work for me.
But it brings another question, that maybe you could clarify: what is the point of reference for forces I pass to `InverseDynamics` method? Mpc gives me forces applied to robot's body (center of mass of body), are they accurate for this function? I'm also aware that I must compute whole spatial vector.

Thanks in advance once again for your support and time!

Kwach00, 2022-07-20:
Okay, now this idea is clear to me. But in my case last body in kinematic chain is whole tibia (foot and tibia are merged).  This function can handle that configuration? It would still work properly without feet bodies? 

Thanks again,

Chris

mjhmilla, 2022-07-20:
Yes, it will work in the sense that the wrench you enter will be applied to the body. There is no restriction, in the function InverseDynamics, on what that wrench can be: you can certainly apply a wrench that does not make physical sense for that body.

Kwach00, 2022-07-22:
Thank you @mjhmilla for taking the trouble to answer all my questions, you helped me a lot!
I believe this issue is ready to be closed.

With gratitude,
Chris

closed 2022-07-22 by Kwach00