rbdl/rbdl #57: Doxygen equations don't build: GPL Ghostscript 9.50: Unrecoverable error, exit code 1

mjhmilla, 2022-07-13:
Is anyone else seeing this string of errors when they try to build the doxygen? For example:

Execution stack:
   %interp_exit   .runexec2   --nostringval--   --nostringval--   --nostringval--   2   %stopped_push   --nostringval--   --nostringval--   --nostringval--   false   1   %stopped_push   1990   1   4   %oparray_pop   1989   1   4   %oparray_pop   1977   1   4   %oparray_pop   1833   1   4   %oparray_pop   --nostringval--   %errorexec_pop   .runexec2   --nostringval--   --nostringval--   --nostringval--   2   %stopped_push   --nostringval--
Dictionary stack:
   --dict:731/1123(ro)(G)--   --dict:0/20(G)--   --dict:76/200(L)--   --dict:5/30(L)--   --dict:11/30(L)--
Current allocation mode is local
Current file position is 8386
GPL Ghostscript 9.50: Unrecoverable error, exit code 1
Generating image form_4.png for formula
Error: /undefined in getenv
Operand stack:
   (outfile)

ju6ge, 2022-07-14:
I have not noticed anything like that, does building the documentation fail for you?

I tried building the doc today and it seems to work for me …

ju6ge, 2022-07-20:
The online documentation is now up to date 😘

closed 2023-11-13 by ju6ge