rbdl/rbdl #58: module 'rbdl' has no attribute 'CalZeroMomentPoint'

timchiu9781, 2022-07-20:
Hi,
I'm now using rbdl with python to implement human pose estimation.
While I need to calculate the zero point, I can not use the **rbdl.CalZeroMomentPoint**.
But I see there has this attribute in the source code.

Did there anything I miss to build when I am doing cmake?

[image]
[image]

Thanks a lot for replying.

ju6ge, 2022-07-20:
Seems like the definition for that function is missing in the cython definitions. It should be fairly straight forward to add this. But I am quite constrained for time. If you need this it should be an easy addition to do yourself. Otherwise I will take a look when I have some time.

ju6ge, 2022-07-20:
Just take a look at the python folder and the crbdl.pxd file ;)

timchiu9781, 2022-07-21:
Thanks for your replying.
I will try to add that function.

timchiu9781, 2022-07-22:
Hello ju6ge,
I have try to fixed the bug.
But it seems not only the cython definitions problem.
But also the calculation of CalZeroMomentPoint missed.

The CalZeroMomentPoint is only shown in rbdl_utils.h, and only define the variable there.
I found that most of the calculation are defined in rbdl_ptr_functions.h, and the CalZeroMomentPoint is not calculated there.
So I can't just add the definitions in cython to fixed it.

Maybe this bug only you can fixed it.
If you have the time, hope you can fix it. Thanks very much.

timchiu9781, 2022-07-22:
I'm not sure that whether I add the CalZeroMomentPoint  function here is right or not. And also the variable defined here.
The variable all comes from rbdl_util.h
 
[image]

ju6ge, 2022-07-22:
You might also need to add the function to rbdl-wrapper.pyx …

I have not touched that code in a while.

timchiu9781, 2022-07-25:
Thanks a lot for your big help.
I have successfully add the function to rbdl-wrapper.pyx.
And also got another error, it seens I miss some definition in some code.
I've try to find out where it is, but I failed.

Do you know what happened to this error?
'CalcZeroMomentPoint' is not a member of 'RigidBodyDynamics'

timchiu9781, 2022-07-25:
I found the problem and solve it successfully.
Now I can use the CalcZeroMomentPoint Func.
Really thanks for your help!

ju6ge, 2022-07-25:
Great \o/ … If you have some time to open a pull request, I would be more than happy to review and include your fixes in the main repository ;)

closed 2023-12-20 by TowoC