rbdl/rbdl #60: Python wrapper - core dumped error

tomerfr, 2022-07-26:
Hello,
I have followed the instructions for compiling RBDL including the python wrapper.
The build process of RBDL passes all of its tests.
The python wrapper is compiled successfully and installed successfully by pip.
Nevertheless, running the "test_wrapper.py" code results in the following error:
"...........Segmentation fault (core dumped)"
Could you please provide me assistance in resolving this issue?
Thank you

ju6ge, 2022-07-26:
I have seen this happen on my arch system as well, only when running the test wrapper though … running a python script that uses rbdl works. Meanwhile on an ubuntu container the test suit works just fine. I can't remember where I got stuck while trying to debug the segmentation fault. It was really obscure and given that the test suite worked on the CI I was using for rbdl-orb I did not follow it further.
So maybe just ignore that the test suit fails weirdly and just use the python wrapper for what ever you want to use it for.

tomerfr, 2022-07-26:
Thank you for the response, but unfortunately I encounter crashes of a python script that uses RBDL, as well.
Specifically, the method "loadModel" crashes ("Aborted (core dumped)") with the exit code 134 (interrupted by signal 6: SIGABRT).
How can I run a python script that uses uses RBDL without getting the "core dumped" error?
Thank you.

ju6ge, 2022-07-26:
okay the plot thickens, the reason I was unable to really debug the testing problem was that i did not now where it occurred. Can you post the model? So I can have a look?

tomerfr, 2022-07-27:
The model is taken from this link:
https://github.com/soshishimada/Neural_Physcap_Demo/tree/master/URDF
Thank you for the help!

MaxReimann, 2022-10-02:
Hi, I am experiencing the same error as @tomerfr - also when using the model from https://github.com/soshishimada/Neural_Physcap_Demo/tree/master/URDF on my Ubuntu 20.04 system. I traced the crash with gdb and it seems to crash on URDF trying to read XML
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7c5d859 in __GI_abort () at abort.c:79
#2  0x00007fffd78ee35a in __cxxabiv1::__terminate (handler=<optimized out>) at /opt/conda/conda-bld/gcc-compiler_1654084175708/work/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48
#3  0x00007fffd78ee3c5 in std::terminate () at /opt/conda/conda-bld/gcc-compiler_1654084175708/work/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:58
#4  0x00007fffd78ee658 in __cxxabiv1::__cxa_throw (obj=<optimized out>, tinfo=0x7fff6fedd428 <typeinfo for urdf::URDFParseError>, dest=0x7fff6feb57a0 <urdf::URDFParseError::~URDFParseError()>)
    at /opt/conda/conda-bld/gcc-compiler_1654084175708/work/gcc/libstdc++-v3/libsupc++/eh_throw.cc:95
#5  0x00007fff6fea6fe4 in urdf::Vector3::fromVecStr(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) [clone .cold] ()
    at /home/max.reimann/anaconda3/envs/physaware/x86_64-conda-linux-gnu/include/c++/11.2.0/ext/new_allocator.h:89
#6  0x00007fff6feb906d in urdf::Transform::fromXml (xml=0x55555944c060) at /home/max.reimann/anaconda3/envs/physaware/x86_64-conda-linux-gnu/include/c++/11.2.0/bits/char_traits.h:329
#7  0x00007fff6fecb53a in urdf::Inertial::fromXml (xml=0x55555944bf80) at /home/max.reimann/rbdl/addons/urdfreader/thirdparty/urdfparser/src/link.cpp:79
#8  0x00007fff6fecc60a in urdf::Link::fromXml (xml=xml@entry=0x55555944be20) at /home/max.reimann/rbdl/addons/urdfreader/thirdparty/urdfparser/src/link.cpp:233
#9  0x00007fff6febcf4c in urdf::UrdfModel::fromUrdfStr (xml_string=...) at /home/max.reimann/rbdl/addons/urdfreader/thirdparty/urdfparser/src/model.cpp:141
#10 0x00007fff6feb434e in RigidBodyDynamics::Addons::URDFReadFromString (
    model_xml_string=0x555559423160 "<robot name=\"dumpUrdf\">\n\t<link name = \"base\">\n\t\t<inertial>\n\t\t\t<origin rpy = \"0 0 0\"  xyz = \"0 0 0\" />\n\t\t\t<mass value =  \"0.0001\" />\n\t\t\t<inertia ixx = \"0.0001\" ixy = \"0\" ixz = \"0\" iyy = \"0.0001\" iyz = "..., model=0x5555592f1d10, floating_base=<optimized out>, verbose=<optimized out>)
    at /home/max.reimann/anaconda3/envs/physaware/x86_64-conda-linux-gnu/include/c++/11.2.0/bits/char_traits.h:329
#11 0x00007fff6feb49a2 in RigidBodyDynamics::Addons::URDFReadFromFile (filename=filename@entry=0x7fff6f1a7410 "./URDF/manual.urdf", model=model@entry=0x5555592f1d10, floating_base=floating_base@entry=true,
    verbose=verbose@entry=false) at /home/max.reimann/anaconda3/envs/physaware/x86_64-conda-linux-gnu/include/c++/11.2.0/bits/basic_string.h:2319
#12 0x00007fff7034210b in rbdl_loadmodel (filename=0x7fff6f1a7410 "./URDF/manual.urdf", model=0x5555592f1d10, floating_base=<optimized out>, verbose=<optimized out>) at /home/max.reimann/rbdl/python/rbdl_loadmodel.cc:33
#13 0x00007fff70342543 in __pyx_pf_4rbdl_48loadModel (__pyx_self=<optimized out>, __pyx_v_kwargs=0x7fff71309f80, __pyx_v_filename=<optimized out>) at /home/max.reimann/rbdl/build/python/rbdl-python.cxx:64677
#14 __pyx_pw_4rbdl_49loadModel (__pyx_self=<optimized out>, __pyx_args=<optimized out>, __pyx_kwds=<optimized out>) at /home/max.reimann/rbdl/build/python/rbdl-python.cxx:64538
#15 0x00005555556d3fa4 in cfunction_call (func=0x7fff70443fb0, args=<optimized out>, kwargs=<optimized out>) at /opt/conda/conda-bld/python-split_1648715631366/work/Objects/methodobject.c:543
#16 0x00005555556965d4 in _PyObject_MakeTpCall (tstate=0x5555559368f0, callable=0x7fff70443fb0, args=<optimized out>, nargs=<optimized out>, keywords=<optimized out>)
    at /opt/conda/conda-bld/python-split_1648715631366/work/Objects/call.c:215
```

I compiled rdbl with ` cmake -D CMAKE_BUILD_TYPE=Debug -D RBDL_BUILD_ADDON_URDFREADER=ON -D RBDL_BUILD_PYTHON_WRAPPER=ON ` in a conda env and installed the python package with pip

ju6ge, 2022-10-02:
Sry I have had a lot of other things on my plate. From the stack trace it seems that the code is not segfaulting in the classical sense meaning it does not look like an memory issue.  
```
0x00007fffd78ee658 in __cxxabiv1::__cxa_throw (obj=<optimized out>, tinfo=0x7fff6fedd428 <typeinfo for urdf::URDFParseError>, dest=0x7fff6feb57a0 <urdf::URDFParseError::~URDFParseError()>)
    at /opt/conda/conda-bld/gcc-compiler_1654084175708/work/gcc/libstdc++-v3/libsupc++/eh_throw.cc:95
```
This line shows that a parsing error is thrown. This means it might be a bug with the [URDF Parser](https://github.com/ORB-HD/URDF_Parser), or some sublte difference of the URDF to most others.

Loading the urdf with `rbdl_urdfreader_util` I get the following error message:
```sh
./addons/urdfreader/rbdl_urdfreader_util manual.urdf 
terminate called after throwing an instance of 'urdf::URDFParseError'
  what():  Parser found 1 elements but 3 expected while parsing vector [0.0]
[1]    66737 IOT instruction (core dumped)  ./addons/urdfreader/rbdl_urdfreader_util manual.urdf
```
Together with the rest of the stack trace that means that in the model file there is a vector that is smaller than expected. Probably some Transform within an inertial within a link description. I hope this helps narrow down the issue.

Kind regards
ju6ge

MaxReimann, 2022-10-04:
Thank you for your reply - I now tried out the model with rbdl v2.6.1 and that seemed to work

closed 2023-11-13 by ju6ge

whl-007, 2024-02-29:
> Thank you for the response, but unfortunately I encounter crashes of a python script that uses RBDL, as well. Specifically, the method "loadModel" crashes ("Aborted (core dumped)") with the exit code 134 (interrupted by signal 6: SIGABRT). How can I run a python script that uses RBDL without getting the "core dumped" error? Thank you.
> 
> 
> 感谢您的回复，但不幸的是，我也遇到了使用RBDL的Python脚本的崩溃。 具体来说，方法“loadModel”崩溃（“Aborted（core dumped）”），退出代码为134（被信号6：SIGABRT中断）。 如何运行一个使用RBDL的Python脚本而不会出现“core dumped”错误？ 谢谢

I have also encountered the problem 'module 'rbdl' has no attribute 'loadModel'. Have you solved this issue? If you could help me solve this problem, I would be greatly appreciated.