rbdl/rbdl #61: Issues with contact constraints and gravity

jcllau, 2022-07-26:
Hi Felix!

A couple hours ago today I made an issue on the pyomeca rbdl-casadi repo and Benjamin (@pariterre ) pointed us to version 3.2.0 (thanks Benjamin!). Menna (@MennatallahRihan) and I compiled 3.2.0 version ourselves, but we (+ Jonathan @jfslin) are still facing the same gravity and contact constraint issues described in the aforementioned issue, which can be found here: https://github.com/pyomeca/rbdl-casadi/issues/22

Below is an example code of a pendulum with a 6-dof base that is free-falling.

`
#include <iostream>
#include "Modeldynamics.h"
#include <rbdl/rbdl.h>

using namespace RigidBodyDynamics;
using namespace RigidBodyDynamics::Math;
RigidBodyDynamics::Model model;
casadi::MX fd(const VectorNd &Q, const VectorNd &Qdot, const VectorNd &Tau)
{
  VectorNd Qddot(model.dof_count);
  ForwardDynamics(model, Q, Qdot, Tau, Qddot);
  return casadi::MX::vertcat({Qdot,Qddot});
}

int main(int argc, char *argv[])
{
  rbdl_check_api_version(RBDL_API_VERSION);

  if (!RigidBodyDynamics::Addons::LuaModelReadFromFile("../models/pendulum.lua", &model, false))
  {
    std::cerr << "Error loading model -- aborting" << std::endl;
    abort();
  }
  cout << "correctly loaded model " << endl;

  auto Q_sym = VectorNd::sym("Q", model.dof_count);
  auto QDot_sym = VectorNd::sym("QDot", model.dof_count);
  auto Tau_sym = VectorNd::sym("Tau", model.dof_count);
  casadi::Function fd_fun = casadi::Function("fd_fun", {Q_sym, QDot_sym, Tau_sym}, {fd(Q_sym, QDot_sym, Tau_sym)}, {"Q", "QDot", "Tau"}, {"QDDot"}).expand();

  auto Q = casadi::MX::zeros(model.dof_count);
  auto QDot = casadi::MX::zeros(model.dof_count);
  auto Tau = casadi::MX::zeros(model.dof_count);

  casadi::MX QDDot = fd_fun(casadi::MXDict{{"Q", Q}, {"QDot", QDot}, {"Tau", Tau}}).at("QDDot");

  // std::cout << QDDot << std::endl;
  // auto x = casadi::MX::vertcat({Q_sym, QDot_sym});

  casadi::MXDict ode_dict = {{"x", casadi::MX::vertcat({Q_sym, QDot_sym})}, {"p", Tau_sym}, {"ode", QDDot}};
  vector<double> x_init = {0, 0, 0, 0, 0, 0,0, 0, 0, 0, 0, 0};
  vector<double> u_init = {0, 0, 0, 0, 0, 0};
  casadi::Function ref_integrator = integrator("ref_integrator",
                                               "cvodes", ode_dict, {{"tf", 0.01}, {"verbose", true}, {"expand", true}});

  map<string, DM> arg, res;

  std::ofstream of("animation.csv");
  for (int i = 0; i < 90; i++)
  {
    if (i < 1)
    {
      arg["x0"] = x_init;
    }
    else
    {
      arg["x0"] = res.at("xf");
    }
    arg["p"] = u_init;
    res = ref_integrator(arg);

    cout << setw(2) << "xf = " << res.at("xf") << endl;
    of << i << ", ";
    for (unsigned int i = 0; i < model.dof_count; i++)
    {
      of << res.at("xf")(i) << ", ";
    }
    of << "\n";
  }

  return 0;
}
`

We are not sure if this is the correct implementation, so any guidance would be greatly appreciated. It would also be helpful if you could provide us with a similar sample code that utilizes integration and ForwardDynamics() or ForwardDynamicsContactsDirect() (we notice that the casadi_simple example only calls FD once).

Hope to hear from you soon!

Thanks,
Menna and Jan

pariterre, 2022-07-26:
If you add "c++" after you ` ` ` it will color things as well :)

pariterre, 2022-07-26:
```c++
casadi::MX fd(const VectorNd &Q, const VectorNd &Qdot, const VectorNd &Tau)
{
  VectorNd Qddot(model.dof_count);
  ForwardDynamics(model, Q, Qdot, Tau, Qddot);
  return casadi::MX::vertcat({Qdot,Qddot});
}
```
If I understood well your problem, you wanted to do constrained forward dynamics, but here you call the non contact dynamics. You should be calling `ForwardDynamicsConstraintsDirect`

pariterre, 2022-07-26:
Another issue (if your problem is the gravity one) may be related to the "lua-reader" as I never tested this one myself. It is possible that the model is simply not loaded properly

jcllau, 2022-07-26:
Re `ForwardDynamicsContactsDirect`: we are simply trying to do a free-fall without constraints as a first step since this isn't working for us either.

As for the lua-reader, we have used it a number of times and it works. Specifically to the gravity, we also explicitly added the line `model.gravity = Vector3d(0., 0., -9.81)` in a different cpp file we have (which is not reflected in the sample code above), and it having that line or not did not make a difference in the behaviour of the model

pariterre, 2022-07-26:
```c++
  auto Q = casadi::MX::zeros(model.dof_count);
  auto QDot = casadi::MX::zeros(model.dof_count);
  auto Tau = casadi::MX::zeros(model.dof_count);

  casadi::MX QDDot = fd_fun(casadi::MXDict{{"Q", Q}, {"QDot", QDot}, {"Tau", Tau}}).at("QDDot");
```

This also seems a little suspicious to me. Why do you relate QDDot to Q, Qdot and Tau which are not MX::sym but actual MX? I'll have a try!

pariterre, 2022-07-26:
Using you code (but loading the model with biorbd instead of LUA returns this)
```
&"warning: GDB: Failed to set controlling terminal: Ioctl() inappropri\303\251 pour un p\303\251riph\303\251rique\n"
can't find linker symbol for virtual table for `std::basic_ofstream<char, std::char_traits<char> >' value
  found `casadi::Sparsity::assign_cached(long long, long long, long long const*, long long const*, bool)' instead
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::init") [.../casadi/core/function_internal.cpp:477]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::create_function odeF:[x, p, t]->[ode]") [.../casadi/core/oracle_function.cpp:132]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::create_function quadF:[x, p, t]->[quad]") [.../casadi/core/oracle_function.cpp:132]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::create_function odeB:[rx, rp, x, p, t]->[rode]") [.../casadi/core/oracle_function.cpp:132]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::create_function quadB:[rx, rp, x, p, t]->[rquad]") [.../casadi/core/oracle_function.cpp:132]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 3.3101e-19, 9.78342e-23, -0.0981, -1.1594e-23, 3.21643e-19, 1.95668e-19]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 6.62019e-19, 1.95668e-22, -0.1962, -2.3188e-23, 6.43285e-19, 3.91337e-19]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 9.93029e-19, 2.93503e-22, -0.2943, -3.4782e-23, 9.64928e-19, 5.87005e-19]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 1.32404e-18, 3.91337e-22, -0.3924, -4.6376e-23, 1.28657e-18, 7.82673e-19]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 1.65505e-18, 4.89171e-22, -0.4905, -5.797e-23, 1.60821e-18, 9.78342e-19]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 1.98606e-18, 5.87005e-22, -0.5886, -6.9564e-23, 1.92986e-18, 1.17401e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 2.31707e-18, 6.84839e-22, -0.6867, -8.1158e-23, 2.2515e-18, 1.36968e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 2.64808e-18, 7.82673e-22, -0.7848, -9.2752e-23, 2.57314e-18, 1.56535e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 2.97909e-18, 8.80508e-22, -0.8829, -1.04346e-22, 2.89478e-18, 1.76101e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 3.3101e-18, 9.78342e-22, -0.981, -1.1594e-22, 3.21643e-18, 1.95668e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 3.6411e-18, 1.07618e-21, -1.0791, -1.27534e-22, 3.53807e-18, 2.15235e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 3.97211e-18, 1.17401e-21, -1.1772, -1.39128e-22, 3.85971e-18, 2.34802e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 4.30312e-18, 1.27184e-21, -1.2753, -1.50722e-22, 4.18135e-18, 2.54369e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 4.63413e-18, 1.36968e-21, -1.3734, -1.62316e-22, 4.503e-18, 2.73936e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 4.96514e-18, 1.46751e-21, -1.4715, -1.7391e-22, 4.82464e-18, 2.93502e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 5.29615e-18, 1.56535e-21, -1.5696, -1.85504e-22, 5.14628e-18, 3.13069e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 5.62716e-18, 1.66318e-21, -1.6677, -1.97098e-22, 5.46793e-18, 3.32636e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 5.95817e-18, 1.76102e-21, -1.7658, -2.08692e-22, 5.78957e-18, 3.52203e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 6.28918e-18, 1.85885e-21, -1.8639, -2.20286e-22, 6.11121e-18, 3.7177e-18]
CasADi - 2022-07-26 17:25:10 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [0, 0, 0, 0, 0, 0, 6.62019e-18, 1.95668e-21, -1.962, -2.3188e-22, 6.43285e-18, 3.91337e-18]
[... cut ...]
xf = [4.58134e-18, 1.35407e-21, -3.88568, -1.60467e-22, 4.45169e-18, 2.70815e-18, 7.01196e-18, 2.07248e-21, -8.7309, -2.45602e-22, 6.81354e-18, 4.14495e-18]
CasADi - 2022-07-26 17:27:22 MESSAGE("ref_integrator::reset") [.../casadi/interfaces/sundials/cvodes_interface.cpp:241]
xf = [4.65244e-18, 1.37509e-21, -3.9735, -1.62957e-22, 4.52079e-18, 2.75018e-18, 7.13821e-18, 2.10979e-21, -8.829, -2.50024e-22, 6.93621e-18, 4.21958e-18]
``` 
which seems to be the expected results if the time is about half a second and the gravity is on Z

MennatallahRihan, 2022-07-26:
Upon changing the fd function as you mentioned above, these are the results we get.  Although it's only a free fall in the Y direction, the other degrees of freedom seem to be changing at a much bigger rate than your output.
step | TX |TY |TZ | RX | RY | RZ
-- | -- | -- | -- | -- | -- | --
0 | -1.78002E-07 | -0.000544207 | 6.76824E-06 | 2.06705E-07 | -1.80459E-06 | 5.00003E-06
1 | -1.43072E-06 | -0.00219254 | 3.38692E-05 | 1.62499E-06 | -1.44371E-05 | 2.00006E-05
2 | -4.85565E-06 | -0.00497065 | 9.0874E-05 | 5.38925E-06 | -4.87648E-05 | 4.50027E-05
3 | -1.15795E-05 | -0.00890729 | 0.000186534 | 1.25535E-05 | -0.000115739 | 8.00095E-05
4 | -2.27619E-05 | -0.0140346 | 0.000328779 | 2.40937E-05 | -0.000226431 | 0.000125027
5 | -3.96002E-05 | -0.0203884 | 0.000524711 | 4.0911E-05 | -0.000392077 | 0.000180066
6 | -6.33339E-05 | -0.0280086 | 0.000780607 | 6.38341E-05 | -0.00062412 | 0.00024514
7 | -9.52504E-05 | -0.0369397 | 0.00110192 | 9.36224E-05 | -0.000934261 | 0.000320272
8 | -0.00013669 | -0.0472315 | 0.00149326 | 0.000130969 | -0.00133451 | 0.000405488
9 | -0.000189049 | -0.0589394 | 0.00195842 | 0.000176502 | -0.00183722 | 0.000500826
10 | -0.000253798 | -0.0721256 | 0.00250036 | 0.000230794 | -0.00245525 | 0.000606329
11 | -0.000332477 | -0.0868598 | 0.00312123 | 0.000294358 | -0.00320192 | 0.000722055
12 | -0.000426712 | -0.10322 | 0.00382235 | 0.000367657 | -0.00409116 | 0.00084807
13 | -0.000538219 | -0.121296 | 0.00460428 | 0.000451103 | -0.00513759 | 0.000984453
14 | -0.000668821 | -0.141187 | 0.00546679 | 0.000545071 | -0.00635663 | 0.0011313
15 | -0.000820457 | -0.163007 | 0.00640898 | 0.000649896 | -0.00776464 | 0.00128872
16 | -0.000995198 | -0.186885 | 0.00742931 | 0.000765886 | -0.009379 | 0.00145684
17 | -0.00119526 | -0.212972 | 0.00852571 | 0.000893329 | -0.0112184 | 0.0016358
18 | -0.00142305 | -0.241438 | 0.00969578 | 0.0010325 | -0.0133029 | 0.00182577
19 | -0.00168113 | -0.272485 | 0.0109369 | 0.00118368 | -0.0156543 | 0.00202693
20 | -0.00197231 | -0.306343 | 0.0122468 | 0.00134717 | -0.0182962 | 0.0022395
21 | -0.00229965 | -0.343287 | 0.0136236 | 0.00152331 | -0.0212548 | 0.00246371
22 | -0.00266651 | -0.383643 | 0.0150667 | 0.00171248 | -0.0245588 | 0.00269982
23 | -0.00307658 | -0.427798 | 0.0165776 | 0.00191518 | -0.0282401 | 0.00294814
24 | -0.00353397 | -0.47622 | 0.0181611 | 0.00213201 | -0.0323346 | 0.00320898
25 | -0.00404328 | -0.529484 | 0.0198269 | 0.00236378 | -0.0368828 | 0.00348271
26 | -0.00460967 | -0.588297 | 0.0215922 | 0.00261151 | -0.0419312 | 0.00376973
27 | -0.00523905 | -0.653548 | 0.0234855 | 0.00287658 | -0.0475331 | 0.00407049
28 | -0.00593818 | -0.726371 | 0.0255515 | 0.00316078 | -0.053751 | 0.00438548
29 | -0.00671497 | -0.808241 | 0.0278598 | 0.00346655 | -0.060659 | 0.00471526
30 | -0.00757876 | -0.90112 | 0.0305172 | 0.00379714 | -0.0683462 | 0.00506043
31 | -0.00854078 | -1.00769 | 0.0336879 | 0.00415703 | -0.0769218 | 0.00542165
32 | -0.0096149 | -1.13177 | 0.0376265 | 0.00455238 | -0.0865234 | 0.00579969
33 | -0.0108187 | -1.27896 | 0.0427355 | 0.00499194 | -0.0973289 | 0.00619537
34 | -0.0121753 | -1.45808 | 0.04967 | 0.00548849 | -0.109578 | 0.00660962
35 | -0.0137167 | -1.68402 | 0.0595443 | 0.00606158 | -0.12361 | 0.0070435
36 | -0.0154911 | -1.98511 | 0.0743864 | 0.00674321 | -0.139946 | 0.0074982
37 | -0.0175796 | -2.42655 | 0.0983048 | 0.00759243 | -0.15949 | 0.0079751
38 | -0.0201574 | -3.23102 | 0.141342 | 0.00874738 | -0.18424 | 0.00847584

ju6ge, 2022-07-26:
Hey guys,

I remember that I had an open issue on rbdl-orb which tracked addon support for casadi (I probably should open that issue here again). 

All addons of rbdl are not yet tested with rbdl-casad! The rbdl-addons link against the normal version of rbdl. While rbdl-casadi is its own shared library. Basically every addon that wants to use rbdl-casadi needs to be build and linked against the casadi version. This creates a casadi version of every addon. If this is not done it could lead to all sorts of weird behavior, if you call a rbdl casadi function with a model that was generated from an addon that assues the normal rbdl.
So if you want to use cpp rbdl with casadi and use a model loaded via a file, you first need to add a cmake script to build the casadi version of the addon!

MennatallahRihan, 2022-07-26:
Hey Felix, 

We have a forked version of the current rbdl with the changes required to make the rbdl-luamodel add-on work with the casadi backend. [here](https://github.com/MennatallahRihan/rbdl) we tried it a couple of times and it seems to be loading properly for us.

ju6ge, 2022-07-26:
Well from a quick look that seems like the changes necessary to make it work ;) … If you have some time create a pull request for it ;)

I am not a heavy user of rbdl and even less so with casadi, so I am not sure that I can help here. Happy debugging I guess …

pariterre, 2022-07-26:
Here is a snippet using biorbd (it is easier for me to use my own inferface to rbdl, but it should not change the outcome), based on your code (notice the simplifications QDDot, since you were creating useless casadi functions. 

```c++
#include "biorbd.h"

///
/// \brief main Get and print the generalized acceleration of a model
/// \return Nothing
///
/// This examples shows how to
///     1. Load a model
///     2. Position the model at a chosen position (Q), velocity (Qdot)
///     3. Compute the generalized acceleration (Qddot) assuming a set of generalized forces (forward dynamics)
///     4. Print them to the console
///
/// Please note that this example will work only with the Eigen backend
///

using namespace BIORBD_NAMESPACE;
BIORBD_NAMESPACE::Model model = BIORBD_NAMESPACE::Model("pendulum.bioMod");

casadi::MX fd(const BIORBD_NAMESPACE::utils::Vector &Q, const BIORBD_NAMESPACE::utils::Vector &Qdot, const BIORBD_NAMESPACE::utils::Vector &Tau)
{
  BIORBD_NAMESPACE::utils::Vector Qddot(model.dof_count);
  std::cout << Qddot.size() << std::endl;
  ForwardDynamics(model, Q, Qdot, Tau, Qddot);
  return casadi::MX::vertcat({Qdot, Qddot});
}

int main()
{
    model.setGravity(BIORBD_NAMESPACE::utils::Vector3d(0, 0, -9.81));

    auto Q_sym = BIORBD_NAMESPACE::utils::Vector::sym("Q", model.dof_count);
    auto QDot_sym = BIORBD_NAMESPACE::utils::Vector::sym("QDot", model.dof_count);
    auto Tau_sym = BIORBD_NAMESPACE::utils::Vector::sym("Tau", model.dof_count);

    // std::cout << QDDot << std::endl;
    casadi::MXDict ode_dict = {{"x", casadi::MX::vertcat({Q_sym, QDot_sym})}, {"p", Tau_sym}, {"ode", fd(Q_sym, QDot_sym, Tau_sym)}};
    casadi::Function ref_integrator = integrator("ref_integrator",
                                                 "cvodes", ode_dict, {{"tf", 0.01}, {"verbose", true}, {"expand", true}});

    std::map<std::string, casadi::DM> arg, res;
    std::vector<double> x_init = {0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0};
    std::vector<double> u_init = {0, 0, 0, 0, 0, 0};
    std::ofstream of("animation.csv");
    for (int i = 0; i < 90; i++)
    {
      if (i < 1)
      {
        arg["x0"] = x_init;
      }
      else
      {
        arg["x0"] = res.at("xf");
      }
      arg["p"] = u_init;
      res = ref_integrator(arg);

      std::cout << std::setw(2) << "xf = " << res.at("xf") << std::endl;
      of << i << ", ";
      for (unsigned int i = 0; i < model.dof_count; i++)
      {
        of << res.at("xf")(i) << ", ";
      }
      of << "\n";
    }

    return 0;
}
[... 182 lines cut ...]
68, 0, -2.3355, 0, 0, 0, 0, 
69, 0, -2.40368, 0, 0, 0, 0, 
70, 0, -2.47285, 0, 0, 0, 0, 
71, 0, -2.543, 0, 0, 0, 0, 
72, 0, -2.61413, 0, 0, 0, 0, 
73, 0, -2.68625, 0, 0, 0, 0, 
74, 0, -2.75934, 0, 0, 0, 0, 
75, 0, -2.83342, 0, 0, 0, 0, 
76, 0, -2.90847, 0, 0, 0, 0, 
77, 0, -2.98451, 0, 0, 0, 0, 
78, 0, -3.06153, 0, 0, 0, 0, 
79, 0, -3.13953, 0, 0, 0, 0, 
80, 0, -3.21851, 0, 0, 0, 0, 
81, 0, -3.29847, 0, 0, 0, 0, 
82, 0, -3.37941, 0, 0, 0, 0, 
83, 0, -3.46134, 0, 0, 0, 0, 
84, 0, -3.54425, 0, 0, 0, 0, 
85, 0, -3.62813, 0, 0, 0, 0, 
86, 0, -3.713, 0, 0, 0, 0, 
87, 0, -3.79885, 0, 0, 0, 0, 
88, 0, -3.88568, 0, 0, 0, 0, 
89, 0, -3.9735, 0, 0, 0, 0, 
```

Conclusion: there is nothing wrong with your code as is. So I would look at the actual model...

closed 2023-10-01 by ju6ge