rbdl/rbdl #63: About dependency to Boost library in vcpkg

SungChiCHIANG, 2022-08-01:
Hi, my apologies if this question is not appropriate here.

When I installed rbdl via vcpkg, I noticed that the whole Boost library is also installed as dependencies. But I thought Boost is optional if you do not need to build the samples? And the package folder do not include the built sample executables which makes me wonder if the installation of Boost is still necessary. It took me almost an hour to finish the installation. It would be nice if the installation of Boost can be skipped.

ju6ge, 2022-08-01:
Boost is required for the URDF Reader addon ;) … though I am not entirely sure if we would actually need all boost packages, probably a subset would be sufficient. But that would require adjusting the RBDL build files. 

You are welcome to try to optimize the build time ;). Though the benefit is limited, because once the dependencies are build, you do not have to rebuild them. And on systems other than windows, you can just install the binary versions of boost and don't have to deal with the build time.

SungChiCHIANG, 2022-08-01:
Thanks for replying! Since the question is answered, I will close this issue.

closed 2022-08-01 by SungChiCHIANG