rbdl/rbdl #71: Python version issue

Theo-Cheynel, 2022-11-21:
Hi, I have an issue when building rbdl bindings for Python 3.7.

I did the following :
```
# Creating conda environment
conda create -n rbdl python=3.7

git clone https://github.com/rbdl/rbdl.git

# Installing python modules
pip install numpy scipy cython matplotlib

# Installing Catch2
git clone --branch v2.x https://github.com/catchorg/Catch2.git
cd Catch2
cmake -Bbuild -H. -DBUILD_TESTING=OFF
sudo cmake --build build/ --target install

cd ../rbdl/
mkdir rbdl-build
cd rbdl-build/
ccmake ..
# Changed flag `RBDL_BUILD_PYTHON_WRAPPER` to ON, then filled `PYTHON` with `/home/ubuntu/anaconda3/envs/rbdl/bin/python3` and filled `PYTHON_NUMPY_3_INCLUDE_DIR` with `/home/ubuntu/anaconda3/envs/rbdl/lib/python3.7/site-packages/numpy/core/include`
# then pressed c to configure, then g to generate
sudo make

pip install -e python/
export PYTHONPATH=$PYTHONPATH:/home/ubuntu/rbdl/rbdl-build/python
```

When I run `import rbdl` in a Python console, here is the output :
 ```
Python 3.7.15 (default, Nov  7 2022, 22:00:21) 
[GCC 11.2.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import rbdl
/home/ubuntu/anaconda3/envs/rbdl/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: compiletime version 3.8 of module 'rbdl' does not match runtime version 3.7
  return f(*args, **kwds)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "type.pxd", line 9, in init rbdl
ValueError: builtins.type size changed, may indicate binary incompatibility. Expected 880 from C header, got 864 from PyObject
```

It looks like, first of all, RBDL expects Python 3.10 (which I cannot use for other reasons). Does anyone know how to solve this issue ?

My environment :
- Ubuntu 20.04
- Anaconda
- Python 3.7.15

Theo-Cheynel, 2022-11-21:
Apparently this is because my default Python installation (at /usr/bin/python) is Python 3.8...
I tried on a virtual machine with Python 3.10 as the base installation and sure enough, the error now says `compiletime version 3.10`.

I'll be using a Docker image instead

traversaro, 2022-11-21:
Random guess: the cmake build of rbdl is compiling against the system Python as it searches for `python3` in https://github.com/rbdl/rbdl/blob/b6c8acf8e585bce02d90a16f9495014d76764f54/python/CMakeLists.txt#L8, but the conda python is installed just as `python`. Switching to the find packages provided by CMake to find python (such as https://cmake.org/cmake/help/latest/module/FindPython3.html) should ensure that the conda's python is properly found.

ChangjianLi, 2023-06-16:
why I got error 
**export PYTHONPATH=$PYTHONPATH:/home/ubuntu/rbdl/rbdl-build/python**
when I tried to import rbdl using python ? the python version is 3.7
 
[image]

lambdalemon, 2024-08-19:
I ran into the same problem and fixed this by using the build tools provided by conda, i.e. before cmake do
```
conda install conda-build cmake cxx-compiler boost-cpp
```