rbdl/rbdl #72: Joint Angle limits

tmayoff, 2022-11-30:
I'm trying to look for an IK library that's able to take joint limits into account, (for example following limits of a human arm). This seemed promising however I'm struggling to understand the documentation. Are joint limits possible? If so how would I add them.

The `JointConstraitSet` seems like the answer but reading the documentation it seems like it's for constraining, for example the hand, to a certain position/orientation but can't constrain, for example the elbow, to a 'normal' human range. 

Any clarifications you could provide would be very helpful, Thanks

ju6ge, 2022-12-02:
@mjhmilla I think this is a question that you are the most qualified to answer ;)

im-renpei, 2023-06-12:
Hi @tmayoff , have you found a solution to this problem?