rbdl/rbdl #77: Featherstone: Algorithm for G and g

iportolesrb, 2023-02-07:
Hi,

Is there any way to obtain the G matrix of the 8.8 Featherstone's? I didn't find anything in the code but maybe I am missing something. In other words, I see that I can get the implicit constraint matrix (CS.G in your code) but I didn't find the explicit one.

Thank you!

ju6ge, 2023-02-07:
@mjhmilla, @martinfelis  this is a question for you guys 😉

mjhmilla, 2023-02-07:
Dear @iportolesrb,

I've taken a bit of a look into this (after some years of not looking at the constraint code). Our CS.G is K in Eqn. 8.36. Looking at the variable documentation in Constraints.h, the closest thing I can see to G that appears in Sec. 8.8 appears in ForwardDynamicsConstraintsNullSpace (in Constraints.h and Constraints.cc - line 1149). However, I don't see an explicit calculation of the G that appears in Sec. 8.8. We might not be evaluating G in Sec. 8.8.

@martinfelis coded the sparse and nullspace methods (though it was a while ago!). He might be able to better answer your questions if what I've written doesn't help.

iportolesrb, 2023-02-15:
Hi,

Thank you for your answers. So I tried to understand the ForwardDynamicsConstraintsNullSpace and build my own G, but at the end it was easier for me to use the method described in 8.11 of Featherstone. I would have liked to have an automated method, as the one that I finally used is not general and is different for every system but I think it's the best option by now.

Thanks!

ju6ge, 2023-02-16:
We are happy to accept pull requests ;) … That is if you have the time and would like to contribute.

iportolesrb, 2023-02-16:
@ju6ge if I have time I will definitely :)