rbdl/rbdl #84: Fixed bodies are not properly considered for the internal computations

saikishor, 2023-06-21:
Hello!

I'm opening this issue in regard to the discussion we had in the PR: https://github.com/rbdl/rbdl/pull/81, that the fixed links are not properly handled internally. 

We could discuss here and come up with a solution for the existing problem.

ju6ge, 2023-06-25:
So if I remember the discussion correctly, the proposed solution to this is to recompute the inertia matrix of the parent body which has fixed bodies attached to it. Probably this should be handled by `rbdl` internally. I think it would be best to do this the moment a fixed body is attached. This way when attaching multiple fixed bodies it would always update the previous inertia matrix. 

@pariterre, @mjhmilla, @martinfelis, @Ipuch, @saikishor Do you see any problem with this? Or other use cases that are not covered?

I would think that the best course of action would be to create a (or some) simple test cases. Maybe some kind of pendulum which highlights the problem. Then a fix could be devised in order to achieve the desired behavior.

Any further thoughts are greatly appreciated. Also I should mention, I don't have time to deal with this (may day job is not in robotics anymore). But I am here to assist and advise as best I can!

saikishor, 2023-06-25:
> So if I remember the discussion correctly, the proposed solution to this is to recompute the inertia matrix of the parent body which has fixed bodies attached to it. Probably this should be handled [...]

I think yes, this should totally fix it. I also believe this is a proper way to do. Right now, the adding fixed bodies, increases the root link mass. I don't know how it is related, but this might need to be changed as well.

ju6ge, 2023-10-01:
Closed as of #87

closed 2023-10-01 by ju6ge