rbdl/rbdl #86: Updating body mass at runtime

pFernbach, 2023-08-16:
Hello, in one of my use case with RBDL I would like to be able to modify the mass of some bodies at runtime, after parsing the urdf. 

The first issue that I encountered was that the model_data_.I and Ic matrices are not updated automatically if I modify a body mass. I had to add this method in my code, and call it with the id of the body I modified:

 ```
 const RigidBodyDynamics::Body &body(rbdl_model_.mBodies[id]);
 RigidBodyDynamics::SpatialRigidBodyInertiad rbi =
      RigidBodyDynamics::SpatialRigidBodyInertiad::createFromMassComInertiaC(
          body.mMass, body.mCenterOfMass, body.mInertia);
  rbdl_model_.getModelData()->Ic[id] = rbi;
  rbdl_model_.getModelData()->I[id] = rbi;
```

First point is: would it be interesting to you to add this method directly in `RigidBodyDynamics::Model` ? 

Second issue: in most of my use cases, I will modify the mass of a fixed body. However, the mass/inertia of the fixed bodies are not considered at all in rbdl computation as they are merged in the movable parent inertia when the body is attached and never updated after that. (Kind of related to https://github.com/rbdl/rbdl/issues/84). 
I made a method that update the mass/inertia of a body according to changes in the fixed bodies attached to it. However, as the "initial" mass and inertia of this body (ie. the one defined in the urdf or at the creation of the model) is overwritten as soon as a fixed body is attached to it, it requires to save this initial values somewhere. 

I had to add cumbersome code in my package in order to do that, and it could be made much easier with a small change in rbdl: saving the initial value of mass and inertia in a body separately from the one that will be updated when a fixed body is added. See https://github.com/rbdl/rbdl/compare/master...pFernbach:rbdl:save_body_inertia?expand=1

Then only issue with this changes is that they break ABI of rbdl. 

Would you be interested that I move this code from my user package directly to rbdl ?

ju6ge, 2023-08-18:
From my point of view, I see no issue with adding your updates. But I am not a heavy user of `rbdl`. I am an maintainer angel, that helps get things done in this repository. RBDL has always been further developed by the people using it. I am here to help make that happen. My general approach is to create as little breakage as possible: 

- So things like function signatures should not change if there is not a very good reason for it. Code should keep compiling when upgrading to a new version. (There might be exceptions, but I have not encountered them yet)
- Adding things in structures is not opposed to that goal!
- Binary compatibility between versions is not a requirement.
- I don't have the impression that the memory footprint of rbdl is too big (but that might just be me in my bubble). So adding more things is not inherently something I am opposed to.

So from my side, I would be happy if you want to contribute your changes so that everybody can benefit from them. So thanks for reaching out and offering you help.

Lastly what do other people using `rbdl` think? @pariterre, @martinfelis , @mjhmilla @saikishor

ju6ge, 2023-08-18:
One thing I am not sure about if it really necessary to keep the original inertial and mass values. Why not compute a matrix to reverse the changes and then update them again? The previous values of the attached body should still be there before the update. Otherwise the logic would get even more messy if multiple fixed bodies are attached to a singe one.

pariterre, 2023-08-18:
We definitely would enjoy having a separate body AND merged body API, for similar reasons. So I am for this kind of changes ;)

saikishor, 2023-08-18:
@pFernbach and myself work together, for us having this kind of API is very interesting. We would like to add a body and be able to update the mass and inertia of the bodies. For instance, like attaching a mass to the robot and be able to work on the fly.

May be an API like `updateBodyMass` or `updateBodyInertia` or `updateBodyProperties` would be very interesting. Ideally, this should update the overall model. This could be a separate PR on itself.

Thank you,

mjhmilla, 2023-08-19:
Dear @pFernbach ,

Just to double check, do you want to update a model's mass at run time 

1. to simulate a continuous change of mass (similar to a rocket in which its mass drops as it burns through fuel), 
2. or are you interested in making a mass change at runtime to a system with a fixed set of masses? 

I think that you're interested in case 2. If I'm wrong and you're really interested in case 1, then we'd actually need to modify the equations of motion. Doable, but more involved than evaluating new body inertias.

pFernbach, 2023-08-27:
Thanks for the answers, 
Dear @mjhmilla, we are intersted in the second option. It will be used to update the inertia of an end effector everytime it is grabing or releasing an object. 

@ju6ge thank you for the comment, I will check how to compute the matrix to revert the changes (basically the opposite effect as the Join method). If it leads to the same results I will use this method in order to avoid adding data in the Body struct.

ju6ge, 2023-10-01:
Closed as of #87

closed 2023-10-01 by ju6ge

mjhmilla, 2023-10-01:
Dear @pFernbach,

Given that you're interested in option 2, you can model that end effector-object-interation as a constaint. At the transition between constraint phases you can calculate the impulse so that the constraint is satisfied at the position and velocity level (appropriate for plastic contact) just prior to switching from, for example, an empty phase to the grabbing phase.

To make this a little clearer, lets assume that you're integrating forwards in time and you want to simulate the end effector picking up something.

1. 'empty' phase
- Lets say this system contains no constraints, or some unrelated to the end effector
- Integrating forwards in time using ForwardDynamics 
- You use this until the end effector touches the payload. The constraint grabbing provides functions calcPositionError and calcVelocity error that you can use to monitor the constraint error (without actually enforcing the function).

2. Transition between empty phase to the grabbing phase
a. Stop the integrator and get the state.
b. Update the constraint set to 'grabbing' constraint set to evaluate the new generalized velocities for both objects.
c. Use ComputeConstraintImpulsesNullSpace() to evaluate the new gereralized velocities post-contact.
d. Update the state vector with the new generalized velocities.

3. Continue integrating forwards in time using ForwardDynamicsConstraintsNullSpace

To build the 'grabbing' constraint set I'd recommend using 6 AddLoopConstraints to simulate a weld joint. This is a very simple representation of an end-effector--to--payload-interaction, but it is a good place to start. 

Hope that helps,

Matt