rbdl/rbdl #88: Get different result for calculate Jacobian by Python and C++

jun951113, 2023-09-15:
Hi Author,

Greeting! I have some problem for calculate Jacobian in C++ and Python. C++: CalcPointJacobian and Python: rbdl.CalcPointJacobian, I get different result for using totally same parameters. Like some of them are zero in Python but other value in C++. Could you please help me solve this problem?

ju6ge, 2023-09-15:
Hi,

I can see that you are new to this. So here are some tips on how to open issues in a way that is helpful and enables a useful response. Because you have given me zero information that could be useful in finding out where the issue is! If I sound annoyed that is because, this happens a lot and it is exhausting having to tell people that if the want help from people who offer their time up for free, they should at the very least also take the time to provide all information that is required to reproduce the issue.

So if you open an issue:
- provide a code snippet: in this case python and c++
- provide all files required for the code to run: in this case model files
- be clear about the versions of the tools you are using! c++ version, cython version and so on … often issues arise from using to old versions!

Kind regards,
ju6ge

jun951113, 2023-09-18:
> Hi,
>
> I can see that you are new to this. So here are some tips on how to open issues in a way that is helpful and enables a useful response. Because you have given me zero information that could be [...]

Hi,

I am so sorry I provide unclear question last time.
Here is my code and result: for calculate the rbdl.NonlinearEffects:
Here is my python code with result:
[image]
[image]
[image]

and Here is my c++ code with result:
[image]
[image]

Like the photo I provide, I have totally same function: 
rbdl.NonlinearEffects with same q and same q_dot, but I got different result.
Could you please help me with this problem? I am really appreciate about it.
Sorry again for last time I ask really silly question without any details.

ju6ge, 2023-10-01:
Look we are all here doing this in our spare time. If you want help you need to provide your example in way that it is easily reproducible. I do not have the time to try to guess what model you are using what you are computing just to help.

From what you have provided I have no clue what you are trying to accomplish and how you are doing it, there are many parts where something might go wrong. So if you still need help, provide a simple code example in python and in c++ so that i can verify the issue. The code samples should be self contained meaning all information that is required to run it is there. Even better it should be a simple case just enough to show what is going wrong. Otherwise I am closing this issue because it is unclear that the issue is with the library and not how you are using it.

closed 2023-10-01 by ju6ge