rbdl/rbdl #89: Confused: Dynamics Non-Linear Effects

kklankers, 2023-09-28:
Hey there, 

i am starting to use RBDL. I am quite familiar with robot dynamics but I was confused by the documentation. I am implementing different controllers in pybullet using the python bindings and urdf addon. 

Basically I want to have: tau = M*ddq + C(q,dq) + G(q). 

I used the CompositeRigidBodyAlgorithm to get M, and I wanted to get the Coriolis term via NonlinearEffects. I thought I still need the to compute the gravity term ( I did with InverseDynamics and putting  dq and ddq to zero) , because the documentation says about the "NonlinearEffects" function: 

/** \brief Computes the coriolis forces
 *
 * This function computes the generalized forces from given generalized
 * states, velocities, and accelerations:
 *   \f$ \tau = N(q, \dot{q}, f_\textit{ext}) \f$
 *
 * \param model rigid body model
 * \param Q     state vector of the internal joints
 * \param QDot  velocity vector of the internal joints
 * \param Tau   actuations of the internal joints (output)
 * \param f_ext External forces acting on the body in base coordinates (optional, defaults to NULL)
 */

Am I missing something? Because I observe an error, when putting C(q,dq) and G(q) on the robot. 

I found in the issue sections, that  NonlinearEffects calculates both gravity and coriolis combined. Just putting C(q,dq) in my control law fixes the problems i faced and everything works like expeceted. 

Thanks for your effort and this amazing library !

mjhmilla, 2023-10-01:
Dear @kklankers,

Just to confirm your suspicion, the gravity term is indeed included in the NonlinearEffects function. You can see this by opening Dynamics.cc and scrolling to the NonlinearEffects function: the first lines of code add gravity to the system and then perform the calculation to evaluate the nonlinear effects. I imagine that @martinfelis included the gravity terms in the NonlinearEffects function because its such a trivial amount of code. However, your experience shows that this is a bit confusing.

@martinfelis, we could update the NonlinearEffects function so that it really does match Featherstone's equations one-to-one. Or we could improve the documentation. Given the spirit of RBDL is to be as fast as possible, I'm leaning towards improving the documentation. What are your thoughts?

Matt

martinfelis, 2024-01-20:
Hi @kklankers (and @mjhmilla!),

the function NonlinearEffects() was derived from InverseDynamics and therefore includes the gravity term. By doing so it very efficiently propagates gravity throughout the tree.

@kklankers easiest solution to support robots that already compensate the gravity term would be to set Model::gravity to zero as that is used for the virtual body at index 0.

Best regards,
Martin

mjhmilla, 2024-01-21:
Good point, @martinfelis.