rbdl/rbdl #93: Error when using ‘RBDL-Casadi+URDFReader’ with floating base model and forward dynamics

zhouqq, 2023-11-23:
Hello,
**Issue Description:**
When utilizing the URDFReader from RBDL-Casadi to create a floating base model and then compute the forward dynamics using CasADi for symbolic solutions, an error is encountered. However, when creating a fixed base model, no error occurs.

**Codes to Reproduce:**
```
#include <filesystem>
#include <rbdl/addons/urdfreader/urdfreader.h>
#include <rbdl/rbdl.h>

casadi::MX fd(RigidBodyDynamics::Model &model,
              const RigidBodyDynamics::Math::VectorNd &Q,
              const RigidBodyDynamics::Math::VectorNd &Qdot,
              const RigidBodyDynamics::Math::VectorNd &Tau) {
  RigidBodyDynamics::Math::VectorNd Qddot(model.dof_count);
  ForwardDynamics(model, Q, Qdot, Tau, Qddot);
  return Qddot;
}

int main() {
  std::string source_dir_path =
      std::filesystem::current_path().parent_path().string();
  std::string urdf_path = source_dir_path + "/cartpole_model.urdf";

  RigidBodyDynamics::Model *model = new RigidBodyDynamics::Model();
  // If set 'floating_base=false', there is no error.
  if (!RigidBodyDynamics::Addons::URDFReadFromFile(urdf_path.c_str(), model,
                                                   /*floating_base*/ true,
                                                   /*verbose*/ true)) {
    std::cerr << "[ERROR] Can not load model [" << urdf_path << "]."
              << std::endl;
    abort();
  }

  std::cout << "Dof: " << model->dof_count << std::endl;

  auto Q_sym = RigidBodyDynamics::Math::VectorNd::sym("Q", model->dof_count);
  auto QDot_sym =
      RigidBodyDynamics::Math::VectorNd::sym("QDot", model->dof_count);
  auto Tau_sym =
      RigidBodyDynamics::Math::VectorNd::sym("Tau", model->dof_count);
  casadi::Function fd_fun = casadi::Function(
      "fd_fun", {Q_sym, QDot_sym, Tau_sym},
      {fd(*model, Q_sym, QDot_sym, Tau_sym)}, {"Q", "QDot", "Tau"}, {"QDDot"});

  delete model;

  return 0;
}
```
**Error Message:**
```
terminate called after throwing an instance of 'casadi::CasadiException'
  what():  .../casadi/core/sparsity_internal.cpp:2195: Assertion "in_range(rr, -size1()+ind1, size1()+ind1)" failed:
Out of bounds error. Got elements in range [8,8], which is outside the range [-8,8).
Aborted (core dumped)
```

**Environment:**
Operating System: Ubuntu22.04
Programming Language: cmake 3.22.1
RBDL-Casadi Version: latest master code
CasADi Version: latest master code

Thank you for your assistance in resolving this issue.

ju6ge, 2023-11-23:
Hi thanks for reporting this. Could you provide your model file? It would help trying to reproduce the issue.

@pariterre have you seen something like this before?

pariterre, 2023-11-23:
Hi there!
I have no idea, but to be fair, I never use RBDL directly. We always use our custom parser (biorbd) and most of the time we use it from our python binder. 

Can you provide what is the shape (size?) of Q_sym, QDot_sym and Tau_sym, as well as the dof_count?

zhouqq, 2023-11-24:
Hi, thanks for your reply.
The attachment is the urdf file I used. Please remove the '.txt' since github cannot upload '.urdf' files.
[cartpole_model.urdf.txt](https://github.com/rbdl/rbdl/files/13454342/cartpole_model.urdf.txt)

Using this urdf with fixed base model, Q_sym, QDot_sym, Tau_sym and dof_count is 2. And with floating base model,  Q_sym, QDot_sym, Tau_sym and dof_count is 8. 

I used a humanoid robot urdf model with 24 dof firstly and got a similar error. Then I replaced this simple inverted pendulum model and found that the error still occurred.