rbdl/rbdl #96: rbdl version 3.3.0 compilation of rbdl addon urdfreader failed.

JIANGLI1252, 2024-02-20:
I downloaded the 3.3.0 version of the library and compiled it successfully without addon.
[image]

I need addon urdfreader, so I set up
OPTION (RBDL_BUILD_ADDON_URDFREADER "Build the (experimental) urdf reader" ON)
then compile,  an error apprears as shown in the figure 
[image]
However, this problem did not appear in versions 3.2.1 and 3.2.0. I can compile these two versions successfully
I use ubuntu 20.04.

ju6ge, 2024-05-04:
I tried to reproduce this behavior, but I can compile on ubuntu on 20.04 just fine. How did you obtain the sources? Did you clone the repository recursively?

MaolinLei, 2024-05-07:
I meet the same problem, also I clone the repository recursively. How to solve this problem?

ju6ge, 2024-05-08:
Since I can not reproduce the behavior there is no way for me to tell what is going wrong. My suspicion is that you are missing an important build step, or are missing a system dependency. But cmake should tell you if it can find everything when configuring the project. 

These are the minimal instructions to compile and build rbdl with the urdf reader addon in a bare `ubuntu:20.04` container:
```sh
apt update && apt install git build-essential cmake libeigen3-dev libboost-all-dev
git clone --recursive https://github.com/rbdl/rbdl.git

cd rbdl
mkdir build
cd build

cmake -D RBDL_BUILD_ADDON_URDFREADER=ON ..

make -j 16
```

ju6ge, 2024-08-23:
#102

closed 2024-08-23 by ju6ge