rbdl/rbdl #103: What's the frame of dynamic matrix M, G, C*qdot ?

ytwboxing, 2024-07-08:
For floating base model or other fixed base model, we can use CompositeRigidBodyAlgorithm() to get M, and InverseDynamics() to get G and C*qdot， what's the frame of them, body frame, or world frame?

martinfelis, 2024-07-09:
They are expressed in generalized coordinates. Also sometimes called "reduced coordinates" or joint space. They depend on what kind of joints you use. Each dimensions corresponds to a degree of freedom of a joint.

ytwboxing, 2024-07-09:
thanks for your reply, so for floating base model, dynamic mass inertia M's top left (6, 6) matrix is all expressed in world? or just expressed in base frame(base_link local frame)? and G and C*qdot is as the same as M?