rbdl/rbdl #104: Failed to build when RBDL_BUILD_ADDON_URDFREADER is ON on Ubuntu 18.04

guntgbtgv, 2024-08-04:
```
[ 76%] Built target rbdl
[ 82%] Building CXX object addons/urdfreader/CMakeFiles/rbdl_urdfreader.dir/urdfreader.cc.o
[ 88%] Linking CXX shared library librbdl_urdfreader.so
[ 88%] Built target rbdl_urdfreader
[ 94%] Linking CXX executable rbdl_urdfreader_util
librbdl_urdfreader.so.3.2.0: undefined reference to `urdf::UrdfModel::fromUrdfStr(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
librbdl_urdfreader.so.3.2.0: undefined reference to `urdf::UrdfModel::getLink(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
librbdl_urdfreader.so.3.2.0: undefined reference to `urdf::Rotation::getRpy(double&, double&, double&) const'
collect2: error: ld returned 1 exit status
make[2]: *** [addons/urdfreader/CMakeFiles/rbdl_urdfreader_util.dir/build.make:132: addons/urdfreader/rbdl_urdfreader_util] Error 1
make[1]: *** [CMakeFiles/Makefile2:719: addons/urdfreader/CMakeFiles/rbdl_urdfreader_util.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

```

when I built on ubuntu 20.04 there was no issue. However when I built on ubuntu 18.04, this happened.

I correctly setup submodule by running `git submodule init` and `git submodule update`

guntgbtgv, 2024-08-10:
I re-built everything, including minor modifications of the code, and it worked.

ofir1080, 2024-11-01:
Hi @guntgbtgv,
I and facing the same issue. Can you share the modifications you have to do?
Thanks!

guntgbtgv, 2024-11-03:
@ofir1080  check #85 . If I recall correctly, this was the modifications I made

guntgbtgv, 2024-11-08:
I came across the same issue, on a different machine. This time, re-building everything does not work.

ju6ge, 2024-11-08:
I just tried to reproduce the error. I am running Ubuntu 18.04 via Container:

Here the minimal instructions to build RBDL with URDF Reader support.
```
> podman run -it --rm ubuntu:18.04
$ apt install libboost-dev cmake cmake-curses-gui build-essential libeigen3-dev git
$ git clone --recursive https://github.com/rbdl/rbdl
$ cd rbdl
$ mkdir build && cd build
$ cmake -D RBDL_BUILD_ADDON_URDFREADER=ON ..
$ make -j
```

All of this runs smoothly without error. And since I am running via container this is a clean environment where nothing else is preconfigured. So any Issues you are seeing probably come from things that are different on your system.

guntgbtgv, 2024-11-11:
@ju6ge  The other machine I used this time is actually Ubuntu 20.04, but RTOS
```
kist@kist-NUC13ANKi7:~/LG_ws/thirdparty/rbdl/build$ uname -a
Linux kist-NUC13ANKi7 6.1.61-xenomai #3 SMP PREEMPT_DYNAMIC IRQ_PIPELINE Fri Sep  6 10:50:49 KST 2024 x86_64 x86_64 x86_64 GNU/Linux
```

and error message goes

```
[ 76%] Built target rbdl
[ 88%] Built target rbdl_urdfreader
[ 94%] Linking CXX executable rbdl_urdfreader_util
/usr/bin/ld: librbdl_urdfreader.so.3.3.1: undefined reference to `urdf::UrdfModel::fromUrdfStr(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: librbdl_urdfreader.so.3.3.1: undefined reference to `urdf::UrdfModel::getLink(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: librbdl_urdfreader.so.3.3.1: undefined reference to `urdf::Rotation::getRpy(double&, double&, double&) const'
collect2: error: ld returned 1 exit status
make[2]: *** [addons/urdfreader/CMakeFiles/rbdl_urdfreader_util.dir/build.make:118: addons/urdfreader/rbdl_urdfreader_util] Error 1
make[1]: *** [CMakeFiles/Makefile2:286: addons/urdfreader/CMakeFiles/rbdl_urdfreader_util.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

Any idea how can I resolve above errors?

closed 2025-06-02 by ju6ge